verilator/verilator #5229: Internal fault/error if an enum object is referenced in a top module and a module is instantiated using using a parameter

goekce, 2024-07-05:
`m.sv`:
```sv
module m #(
    int unsigned N = 8
);
  enum {
    A,
    B
  } st;
endmodule
```

`tb.sv`:
```sv
module tb;
  localparam int unsigned N = 8;
  m #(N) dut;

  initial wait (dut.st == 
  `ifndef SECOND_CASE
      dut.A
  `else
      dut.B
  `endif
      );
endmodule
```

### First case:
```sh
verilator --binary -MAKEFLAGS '--silent' --quiet -j 0 m.sv tb.sv
```
Output:
```
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt                             ^
```

### Second case:
```sh
verilator --binary -MAKEFLAGS '--silent' --quiet -j 0 +define+SECOND_CASE  m.sv tb.sv
```
Output:
```
%Error: Internal Error: tb.sv:9:11: ../V3Width.cpp:2425: EnumItemRef can't deref back to an Enum
                                  : ... note: In instance 'tb'
    9 |       dut.B
      |           ^
```

Version 5.026 on ArchLinux

The following cases do not emit an error:
1. Instantiating `m` without using the parameter port (`m dut;`)
2. removing the  hierarchical name `dut.*` by using `dut.st == 0` instead in the `initial.

Does someone have an idea about where this error comes from?

wsnyder, 2024-07-05:
During elaboration, newmodules are made, and this seems to break the reference and give the error you see.  Probably a bug in V3LinkDpt or V3Param.

Those canbehard to understand but might you be interested in fixing?

goekce, 2024-07-05:
I am interested in fixing, but I have other bug issues that I created before which I personally find more important. So others are welcome.