verilator/verilator #5230: How to dump DfgGraph with a flatten design? Can I generate a flatten design of SystemVerilog/Verilog code?

enochliang, 2024-07-06:
I am working on a CPU design (lowRISC IBEX CPU) in SystemVerilog, which has multiple modules under the top in different RTL source code files. And I can successfully generate DfgGraph files by using the command in the previous discussion ([https://github.com/verilator/verilator/issues/5223#issue-2390391963]). But they are separated and placed in different files according to different modules.

I want to get a **_DfgGraph with a flatten design_**, or just generate a **_flatten RTL source code including every submodule_**, and then dump the DfgGraph of it.

So I've try to add `--flatten` flag like this (`verilator --cc --dumpi-dfg 9 --flatten {design.sv} --top {design_name}`), but nothing changes.
I also tryed `verilator --xml-only --flatten {design.sv} --top {design_name}`, and I get the flatten AST successfully, because according to the docs, `--flatten` is usually used with `--xml-only`.

Is it possible for me to get the **flatten DFG** or the **flatten SV** file.

wsnyder, 2024-07-06:
@gezalore

gezalore, 2024-07-06:
There should be a "*024root-postinline-whole-input.dot" with --flaten, that should contain everything. Good luck visualising it though :)

closed 2024-07-06 by gezalore

enochliang, 2024-07-07:
I found it, thanks a lot!