verilator/verilator #5233: Still failing with CPP Error after fix to #5221

deepans88, 2024-07-08:
Hi. We are running into new issues after I tried the fix to #5221 . I changed the structure to packed like this

package  dft_mbist;

  typedef struct packed   {
    logic         BIST;
    logic [38:0]  WEB; // MBIST bit write enable
    logic         CEB;
  } t_i_mbist_top_mem;
Verilator is still failing with this error

**verilated.o: In function VerilatedContext::addModel(VerilatedModel*)': verilated.cpp:(.text+0x4750): undefined reference to std::__cxx11::basic_ostringstream<char, std::char_traits, std::allocator >::basic_ostringstream()'
verilated.o: In function VerilatedContextImp::commandArgVlUint64(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long&, unsigned long, unsigned long)::{lambda(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)#1}::operator()(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const': verilated.cpp:(.text+0x49d8): undefined reference to std::__cxx11::basic_stringstream<char, std::char_traits, std::allocator >::basic_stringstream()'
verilated.o: In function Verilated::scTimePrecisionError(int, int)': verilated.cpp:(.text+0x4e59): undefined reference to std::__cxx11::basic_ostringstream<char, std::char_traits, std::allocator >::basic_ostringstream()'
collect2: error: ld returned 1 exit status
make[1]: *** [Vram_wrapper] Error 1
rm Vram_wrapper__ALL.verilator_deplist.tmp
%Error: make -C obj_dir -f Vram_wrapper.mk -j 32 exited with 2**

deepans88, 2024-07-10:
@wsnyder : Any idea why I am getting this error. Is this related to gcc version

Please find tool version I am using now

**Verilator version :- Verilator 5.027 devel rev v5.026-2-gd5cfe1a
GCC version:-gcc (GCC) 7.3.0
OS type : CentOS Linux 7 (Core)**

wsnyder, 2024-07-10:
Using the example in #5221 but changed to "packed struct" I don't get any failures.  Perhaps you have an old verilator version or didn't completely clean your directory?

The link error looks like a broken build or something. Googling the error suggest your compler or setup is otherwise broken in ways unrelated to verilator.  https://stackoverflow.com/questions/75336475/how-to-resolve-std-cxx11basic-ostringstreamchar-stdchar-traitschar

wsnyder, 2024-07-10:
Also CentOS 7 is very old and already end-of-life, I would strongly suggest something more modern.

deepans88, 2024-07-12:
Thanks. Issue was with a 2nd GCC version. Thanks for your inputs

closed 2024-07-12 by deepans88