verilator/verilator #5237: Too many arguments in function call to TASK, misleading error message.

Kreijstal, 2024-07-08:
I have some code that happens to work with questa and xilinix vivado, but it just does not work with verilator, and the error message seems misleading, however the code is long so I removed a big chunk just to have the most important part that still manages to reproduce it.

https://gist.github.com/Kreijstal/4d0fed1e4e7818d9e14e73355bcb3210

`git clone  https://gist.github.com/4d0fed1e4e7818d9e14e73355bcb3210.git example && cd example`
run with `verilator tb_hello_world.sv -main --timing --binary --trace-fst  dm_pkg.sv`

and you get 
```
%Error: rvlab_tests.sv:21:35: Too many arguments in function call to TASK 'cycle_dr_32'
   21 |     jtag.cycle_dr_32(idcode_read, '0);
      |                                   ^~
        tb_hello_world.sv:16:1: ... note: In file included from 'tb_hello_world.sv'
%Warning-INITIALDLY: jtag_master.sv:35:21: Non-blocking assignment '<=' in initial/final block
                                         : ... note: In instance 'tb_hello_world.jtag'
                                         : ... This will be executed as a blocking assignment '='!
   35 |     ir_stored_valid <= '0;
      |                     ^~
                     tb_hello_world.sv:7:1: ... note: In file included from 'tb_hello_world.sv'
                     ... For warning description see https://verilator.org/warn/INITIALDLY?v=5.026
                     ... Use "/* verilator lint_off INITIALDLY */" and lint_on around source to disable this message.
%Warning-INITIALDLY: jtag_master.sv:36:21: Non-blocking assignment '<=' in initial/final block
                                         : ... note: In instance 'tb_hello_world.jtag'
                                         : ... This will be executed as a blocking assignment '='!
   36 |     ir_stored       <= '0;
      |                     ^~
                     tb_hello_world.sv:7:1: ... note: In file included from 'tb_hello_world.sv'
%Error: Internal Error: tb_hello_world.sv:5:15: ../V3TraceDecl.cpp:439: Expected an interface scope name to have at least one dot
                                              : ... note: In instance 'jtag_master::__VDynScope_2'
    5 |   jtag_master jtag();
      |               ^~~~
```

which makes no sense since cycle_dr_32 accepts exactly 2 arguments...?

rhanqtl, 2024-07-10:
Here is a smaller case which can reproduce the issue:

```systemverilog
module tb;
  bit q;
  intf u_intf();

  initial u_intf.tt(q);
endmodule

interface intf;
  task tt(output q);
    // rewritten to fork-join, and `q` is captured in a "DynScope",
    // see `DynScopeVisitor` in V3Fork.cpp.
    q <= 1'b1;
  endtask : tt
endinterface
```

wsnyder, 2024-07-10:
@kbieganski seems related to delay handling inside interface's tasks.

Kreijstal, 2024-07-11:
> Here is a smaller case which can reproduce the issue:
> 
> ```systemverilog
> module tb;
>   bit q;
>   intf u_intf();
> 
>   initial u_intf.tt(q);
> endmodule
> 
> interface intf;
>   task tt(output q);
>     // rewritten to fork-join, and `q` is captured in a "DynScope",
>     // see `DynScopeVisitor` in V3Fork.cpp.
>     q <= 1'b1;
>   endtask : tt
> endinterface
> ```
> 
> Seems that the port `q` is removed unexpectedly at some place.

Thank you for making a smaller case, I was not sure what the problem was or how important were those things, but yes it seems that might be the problem

kbieganski, 2024-07-11:
Thanks for the minimal test case. It's actually not related to interfaces, as this also breaks:

```systemverilog
module t;
  bit q;

  initial tt(q);

  task tt(output q);
    q <= 1'b1;
  endtask : tt
endmodule
```

And, as @rhanqtl noticed, it gets transformed into this (also breaks):

```systemverilog
module t;
  bit q;

  initial tt(q);

  task tt(output q);
    fork
      q = 1'b1;
    join_none
  endtask : tt
endmodule
```

I think I may have a fix, I'll get back to you.

kbieganski, 2024-07-11:
The fix requires more consideration, so I'll hold off opening a PR for now.

There's another bug here, from the original poster's output:
```
%Error: Internal Error: tb_hello_world.sv:5:15: ../V3TraceDecl.cpp:439: Expected an interface scope name to have at least one dot
                                              : ... note: In instance 'jtag_master::__VDynScope_2'
    5 |   jtag_master jtag();
      |               ^~~~
```
This isn't reproduced by the minimal test case.

Kreijstal, 2024-07-11:
I mean maybe these problem is just 2 bugs after another, and you solved at least one, that is not bad right?

kbieganski, 2024-07-11:
If anyone's interested, the first bug is due to `V3Fork` unlinking non-input ports in a few places like this:
https://github.com/verilator/verilator/blob/570e1bc35a96e0de515de453102a00a10edf161c/src/V3Fork.cpp#L115
But changing it so it doesn't do that may not be sufficient (as in, the output may not be propagated outside the task). Needs some testing, I'll do it when I find the time.

This second bug is there regardless of that (it's already there in your original output). It's separate from the first one, but it's also related to `V3Fork`.

kbieganski, 2024-08-08:
#5343 should fix this.
The other errors:
```
%Warning-INITIALDLY: jtag_master.sv:35:21: Non-blocking assignment '<=' in initial/final block
```
```
%Error: Internal Error: tb_hello_world.sv:5:15: ../V3TraceDecl.cpp:439: Expected an interface scope name to have at least one dot
```
are separate problems.

PR #5343 (2024-08-08): Fix forks capturing non-input ports in tasks
Adds a write-back for outputs/inouts captured in forks. If there's a delay, emits an `UNSUPPORTED` error (how would that even work?).

Fixes #5237
closed 2024-08-08 by wsnyder