verilator/verilator #5238: Support pre-commit via docker image on docker hub

cbachhuber, 2024-07-08:
I'd like to use verilator as a pre-commit hook, as part of https://pre-commit.com. This framework auto-fetches hook binaries from public resources such as pypi or docker hub, but unfortunately not debian/apt.

To support a verilator pre-commit hook with minimal effort, we could publish the [executable image](https://github.com/verilator/verilator/tree/b66fdd7421709fe2cc1c49851d92bfe21019f7ea/ci/docker/run) on docker hub, such that anyone can write their own verilator pre-commit hook. We could even provide a hook as part of this repo (see for example [how hadolint does it](https://github.com/hadolint/hadolint/blob/742b76e3b9b4d45d6716771b0f1fae2fa0069104/.pre-commit-hooks.yaml)), that makes setting up your own verilator hook even easier.

What do you think about this?

wsnyder, 2024-07-08:
There's already Verilator images on docker hub, so I'm not sure what the request is.

Note docker does break occasionally and might even be out-of-date now, so if you want to rely on this we'll need you to be willing to help support docker.

wsnyder, 2024-07-08:
There's already Verilator images on docker hub, so I'm not sure what the request is?

Note docker does break occasionally and might even be out-of-date now, so if you want to rely on this we'll need you to be willing to help support docker.

cbachhuber, 2024-07-09:
Nice! And sorry, I didn't see that. WDYT about https://github.com/verilator/verilator/pull/5241 to avoid this in future?

I'll try to set up a pre-commit hook with this and report back here. You don't need to do anything for now, thank you!

cbachhuber, 2024-07-09:
Works with

```yaml
  - repo: local
    hooks:
      - id: verilator
        name: verilator-lint
        args: [--lint-only]
        language: docker_image
        entry: verilator/verilator:v5.026
        types: ["verilog"]
```

Should we offer this hook as part of this repo? With that, using the hook in other repos would be as simple as

```yaml
  - repo: https://github.com/verilator/verilator
    rev: v5.026
    hooks:
      - id: verilator
```

wsnyder, 2024-07-09:
Probably.  I don't use docker myself - @wallento any comment?

wsnyder, 2024-09-08:
Closing as answered.  @cbachhuber please consider submitting a pull to add your suggested hook.

closed 2024-09-08 by wsnyder

cbachhuber, 2024-09-14:
https://github.com/verilator/verilator/pull/5452