verilator/verilator #5244: <command-line>: fatal error: VMockArrayTestbench__pch.h.fast: No such file or directory

oharboe, 2024-07-09:
Using Verilator 5.027 devel rev v5.026-26-gb66fdd742, running on Ubuntu 23.10.

Tor reproduce:

untar 

```
$ verilator --cc --exe MockArrayTestbench-harness.cpp --assert --coverage-user -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --top-module MockArrayTestbench +define+TOP_TYPE=VMockArrayTestbench -CFLAGS "-O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VMockArrayTestbench -include VMockArrayTestbench.h -fPIC -shared -fvisibility=hidden -Mdir verilated -LDFLAGS -shared -dynamiclib -fvisibility=hidden" --trace -f firrtl_black_box_resource_files.f --bbox-unsup MockArrayTestbench.sv
$ mkdir verilated
$ make -C verilated -f VMockArrayTestbench.mk VMockArrayTestbench
```

Output:

```
[deleted]
<command-line>: fatal error: VMockArrayTestbench__pch.h.fast: No such file or directory
[deleted]
```

b-chmiel, 2024-07-09:
Seems to be related to issue https://github.com/verilator/verilator/issues/5139 resolved by https://github.com/verilator/verilator/pull/5202.

oharboe, 2024-07-09:
I'm running wiht #5202. 

I tried to switch to `--compiler-include`. No luck...

```
$ verilator --cc --exe MockArrayTestbench-harness.cpp --assert --coverage-user -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --top-module MockArrayTestbench +define+TOP_TYPE=VMockArrayTestbench -CFLAGS "-O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VMockArrayTestbench -include VMockArrayTestbench.h -fPIC -shared -fvisibility=hidden -Mdir verilated -LDFLAGS -shared -dynamiclib -fvisibility=hidden" --trace -f firrtl_black_box_resource_files.f --bbox-unsup MockArrayTestbench.sv
```

```
$ make -C verilated -f VMockArrayTestbench.mk VMockArrayTestbench
make: Entering directory '/home/oyvind/OpenROAD-flow-scripts/flow/designs/src/mock-array/foo/verilated'
ccache g++ -Os  -I.  -MMD -I/home/oyvind/verilator/install/share/verilator/include -I/home/oyvind/verilator/install/share/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VMockArrayTestbench -include VMockArrayTestbench.h -fPIC -shared -I'/usr/lib/jvm/java-17-openjdk-amd64/include' -I'/usr/lib/jvm/java-17-openjdk-amd64/include/linux' -fvisibility=hidden   -include VMockArrayTestbench__pch.h.fast -c -o VMockArrayTestbench.o VMockArrayTestbench.cpp
<command-line>: fatal error: VMockArrayTestbench__pch.h.fast: No such file or directory
compilation terminated.
make: *** [/home/oyvind/verilator/install/share/verilator/include/verilated.mk:268: VMockArrayTestbench.o] Error 1
make: Leaving directory '/home/oyvind/OpenROAD-flow-scripts/flow/designs/src/mock-array/foo/verilated'
$ ls verilated/VMockArrayTestbench*pch*
verilated/VMockArrayTestbench__pch.h
verilated/VMockArrayTestbench__pch.h.fast.d
verilated/VMockArrayTestbench__pch.h.fast.gch
verilated/VMockArrayTestbench__pch.h.slow.d
verilated/VMockArrayTestbench__pch.h.slow.gch
```

wsnyder, 2024-07-09:
This is the git master version?  What version of gcc? Try removing all object directories and doing a "ccache -C" between builds.  Also try "./configure CC=clang++" instead.

oharboe, 2024-07-10:
Yes, latest master

```
$ git log -1
commit b66fdd7421709fe2cc1c49851d92bfe21019f7ea (HEAD, origin/master, origin/HEAD, master)
Author: Wilson Snyder <wsnyder@wsnyder.org>
Date:   Mon Jul 8 10:19:51 2024 -0400

    Fix unknown conversion on queues (#5220).
$ git describe 
v5.026-26-gb66fdd742
```

```
$ ccache -C
Clearing... 100.0% [===========================================================]
$ verilator --version
Verilator 5.027 devel rev v5.026-26-gb66fdd742
```

```
verilator --cc --exe MockArrayTestbench-harness.cpp --assert --coverage-user -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --top-module MockArrayTestbench +define+TOP_TYPE=VMockArrayTestbench -CFLAGS "-O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VMockArrayTestbench -fPIC -shared -fvisibility=hidden -Mdir verilated -LDFLAGS -shared -dynamiclib -fvisibility=hidden" --trace -f firrtl_black_box_resource_files.f --compiler-include VMockArrayTestbench.h --bbox-unsup MockArrayTestbench.sv
```

```
$ make -C verilated -f VMockArrayTestbench.mk VMockArrayTestbench
make: Entering directory '/home/oyvind/OpenROAD-flow-scripts/flow/designs/src/mock-array/foo/verilated'
ccache g++ -Os  -I.  -MMD -I/home/oyvind/verilator/install/share/verilator/include -I/home/oyvind/verilator/install/share/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VMockArrayTestbench -include VMockArrayTestbench.h -fPIC -shared -I'/usr/lib/jvm/java-17-openjdk-amd64/include' -I'/usr/lib/jvm/java-17-openjdk-amd64/include/linux' -fvisibility=hidden   -include VMockArrayTestbench__pch.h.fast -c -o VMockArrayTestbench.o VMockArrayTestbench.cpp
<command-line>: fatal error: VMockArrayTestbench__pch.h.fast: No such file or directory
compilation terminated.
make: *** [/home/oyvind/verilator/install/share/verilator/include/verilated.mk:268: VMockArrayTestbench.o] Error 1
make: Leaving directory '/home/oyvind/OpenROAD-flow-scripts/flow/designs/src/mock-array/foo/verilated'
```

oharboe, 2024-07-10:
Tested on a different machine. Same result.

wsnyder, 2024-07-10:
>-include VMockArrayTestbench.h

This is the problem. Debug where this is coming from - verilator wouldn't do this itself, so it's some input that's bad.

b-chmiel, 2024-07-10:
When running on Verilator master (https://github.com/verilator/verilator/commit/11da07d3b90dd5ac4f7369a4002c6d0d8b96209f), clang 18.1.8, after changing paths in the `` and running with the following command:
```
verilator --cc --exe MockArrayTestbench-harness.cpp --assert --coverage-user -Wno-fatal -Wno-WIDTH -Wno-STMTDLY --top-module MockArrayTestbench +define+TOP_TYPE=VMockArrayTestbench -CFLAGS "-O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VMockArrayTestbench -fPIC -shared -fvisibility=hidden -LDFLAGS -shared -dynamiclib -fvisibility=hidden" --trace -f firrtl_black_box_resource_files.f --compiler-include VMockArrayTestbench.h --bbox-unsup MockArrayTestbench.sv -Mdir verilated
```
(note the moved `-Mdir` flag) 

I received different error during the linking stage (which is run after the reported error):
```
mold: error: undefined symbol: main
>>> referenced by /usr/lib64/Scrt1.o:(.text)
>>>               /usr/lib64/Scrt1.o:(_start)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [VMockArrayTestbench.mk:66: VMockArrayTestbench] Error 1
```

Moreover, precompiled headers are present in the `verilated` directory:
```
verilated/VMockArrayTestbench__pch.h.fast.gch
verilated/VMockArrayTestbench__pch.h.slow.gch
```

Also in the `verilated/VMockArrayTestbench__pch.h`, there is an additional include created by `--compiler-include` with provided additional header.

b-chmiel, 2024-07-12:
You're right. The compilation error is connected to incorrect `-LDFLAGS` which are defined inside `-CFLAGS` verilator argument. After changing command to:

```console
verilator --cc --exe MockArrayTestbench-harness.cpp --assert --coverage-user -Wno-fatal -Wno-WIDTH -Wno-STMTDLY -Wno-PINMISSING --top-module MockArrayTestbench +define+TOP_TYPE=VMockArrayTestbench -CFLAGS "-O0 -g -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VMockArrayTestbench -fPIC -shared -fvisibility=hidden" -LDFLAGS "-shared -dynamiclib -fvisibility=hidden" --trace -f firrtl_black_box_resource_files.f --bbox-unsup MockArrayTestbench.sv -Mdir obj_dir --compiler-include VMockArrayTestbench.h
```

Under both Clang 18.1.8 and GCC 14.1.1 (with GNU ld 2.42.0) everything compiles fine.

Note: the `--compiler-include` is redundant, since it only includes verilator-generated file `VMockArrayTestbench.h` which is already included.

Could you please try and check if it helps?

oharboe, 2024-07-12:
I get:
```
$ verilator --cc --exe MockArrayTestbench-harness.cpp --assert --coverage-user -Wno-fatal -Wno-WIDTH -Wno-STMTDLY -Wno-PINMISSING --top-module MockArrayTestbench +define+TOP_TYPE=VMockArrayTestbench -CFLAGS "-O0 -g -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VMockArrayTestbench -fPIC -shared -fvisibility=hidden" -LDFLAGS "-shared -dynamiclib -fvisibility=hidden" --trace -f firrtl_black_box_resource_files.f --bbox-unsup MockArrayTestbench.sv -Mdir obj_dir --compiler-include VMockArrayTestbench.h
- V e r i l a t i o n   R e p o r t: Verilator 5.027 devel rev v5.026-30-g2cfec0ecc
- Verilator: Built from 15.359 MB sources in 194 modules, into 89.476 MB in 49 C++ files needing 0.000 MB
- Verilator: Walltime 25.299 s (elab=2.515, cvt=21.170, bld=0.000); cpu 0.000 s on 1 threads; alloced 1392.637 MB
$ make -C verilated -f VMockArrayTestbench.mk VMockArrayTestbench
make: Entering directory '/tmp/foo/verilated'
ccache g++ -Os  -I.  -MMD -I/home/oyvind/verilator/install/share/verilator/include -I/home/oyvind/verilator/install/share/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -O1 -DVL_USER_STOP -DVL_USER_FATAL -DVL_USER_FINISH -DTOP_TYPE=VMockArrayTestbench -include VMockArrayTestbench.h -fPIC -shared -I'/usr/lib/jvm/java-17-openjdk-amd64/include' -I'/usr/lib/jvm/java-17-openjdk-amd64/include/linux' -fvisibility=hidden   -include VMockArrayTestbench__pch.h.fast -c -o VMockArrayTestbench.o VMockArrayTestbench.cpp
<command-line>: fatal error: VMockArrayTestbench__pch.h.fast: No such file or directory
compilation terminated.
make: *** [/home/oyvind/verilator/install/share/verilator/include/verilated.mk:268: VMockArrayTestbench.o] Error 1
make: Leaving directory '/tmp/foo/verilated'
```

b-chmiel, 2024-07-12:
Could you try `make -C obj_dir -f VMockArrayTestbench.mk VMockArrayTestbench`?

oharboe, 2024-07-12:
> Could you try `make -C obj_dir -f VMockArrayTestbench.mk VMockArrayTestbench`?

Worked! To be precise, I added "-j"...

```
make -j -C obj_dir -f VMockArrayTestbench.mk VMockArrayTestbench
```

oharboe, 2024-07-12:
This problem originated when I was working on Chisel and trying to [run post-synthesis simulation](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2120/commits/7c58bd9066d6ed957a4ec5b004ed0d38a8fb9da4). I'll see if I can understand what goes wrong in Chisel now.

closed 2024-07-12 by wsnyder

oharboe, 2024-07-12:
The fix needs to go into Chiseltest as I understand https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/2120#issuecomment-2225439423

Thank you for all the help!