verilator/verilator #5249: Force on module input / Variable inlining should make this impossible

KlausKoder, 2024-07-11:
## Summery

I was hit by the internal error 'Variable inlining should make this impossible', as in (#5050). Here it seem be caused by using `force` on an module input.

## Example to show the issue

The error happens for me on caliptra-rtl, and is quite easy to reproduce.

```
git clone https://github.com/chipsalliance/caliptra-rtl.git
mkdir -p build && cd build
make -f ../caliptra-rtl/tools/scripts/Makefile TESTNAME=iccm_lock
```

On master `verilator` produce this error:

```
%Error: Internal Error: {}/caliptra-rtl/src/soc_ifc/rtl/wdt.sv:25:45: ../V3DfgPasses.cpp:308: Variable inlining should make this impossible
   25 |     input  logic [WDT_TIMEOUT_PERIOD_W-1:0] timer1_timeout_period,
```

This is triggered by `force`/`release` on the signal from the test-bench:

```
    force caliptra_top_dut.soc_ifc_top1.timer1_timeout_period = 64'hFFFFFFFF_FFFFFFFF;
```

This is pretty standard stuff, i guess it must be an regression caused by commit 5e1fc6e24d.
Unfortunately, the release prior to this commit (v5.020) delibratly throw an error on the attempt to force a module input,
`%Error-ASSIGNIN`.

Verilator versions between 5.016 and v5.020 throw the 

```
%Error-ASSIGNIN: {}/caliptra-rtl/src/integration/tb/caliptra_top_tb.sv:1270:36: Assigning to input/const variable: 'dma_htrans'
 1270 |     force   caliptra_top_dut.rvtop.dma_htrans = 2'b00;

Where {}/caliptra-rtl/src/riscv_core/veer_el2/rtl/el2_veer_wrapper.sv:
  278 | input logic [1:0]                       dma_htrans,
```

I am not sure that it is correct to prohibit force on inputs. I was under the assumption that the whole net would be forced, regardless. As explained in this question [Understand force/release](https://verificationacademy.com/forums/t/understanding-force-and-release/37843)

As far as I can see, there are no *golden* verilator that can build this project:

Version | verilator error on caliptra-rtl
--------------|---------------------------------
v4.228-v5.014  | Confused on very long type-names
v5.016-v5.020  | Refuse forced assign on inputs (Error-ASSIGNIN)
v5.024  |  Assert V3DfgPasses.cpp:314
v5.024-69-g7c9fa8647  | Assert V3DfgPasses.cpp:308
v5.026  |  Assert V3DfgPasses.cpp:308
v5.026-31-g11da07d3b  |  Assert V3DfgPasses.cpp:308

### What 'verilator' command line do we use to run your example?

I used this in my attempt to find a golden version, the options are nothing special.

```
for ver in ~/usr/verilator-*/bin/verilator; do \
  echo Verilator used $ver;\
  $ver ../caliptra-rtl/src/integration/tb/test_caliptra_top_tb.cpp ../caliptra-rtl/src/integration/test_suites/libs/jtagdpi/jtagdpi.c \
  ../caliptra-rtl/src/integration/test_suites/libs/tcp_server/tcp_server.c --cc \
  -CFLAGS "-std=c++17 -I../caliptra-rtl/src/integration/test_suites/libs/jtagdpi/ -I../caliptra-rtl/src/integration/test_suites/libs/tcp_server/" \
  +libext+.v+.sv +define+RV_OPENSOURCE --timescale 1ns/1ps --timing \
  -I. \
  -Wno-WIDTH -Wno-UNOPTFLAT -Wno-LITENDIAN -Wno-CMPCONST -Wno-MULTIDRIVEN -Wno-UNPACKED \
  -f ../caliptra-rtl/src/integration/tb/../config/caliptra_top_tb.vf --top-module caliptra_top_tb \
  -f ../caliptra-rtl/src/integration/tb/../config/caliptra_top_tb.vlt \
  -exe test_caliptra_top_tb.cpp --autoflush  \
  +define+CALIPTRA_INTERNAL_QSPI +define+CALIPTRA_INTERNAL_TRNG +define+CALIPTRA_INTERNAL_UART; \
done
```

`verilator-4228` doens't implement `--timing`.

### Used verilator versions

As previous listed, all 13 tagged versions from v4.228 to v5.026, master and then some.

### OS and distribution

```
$ uname -osr
Linux 6.8.11.fc40.x86_64 GNU/Linux

$ gcc --version
gcc (GCC) 14.1.1 20240522 (Red Hat 14.1.1-4)
```

wsnyder, 2024-07-11:
@gezalore might you investigate, thanks.

KlausKoder, 2024-07-12:
This is an attempt to boil down the error to some essence. The code is now completely non-nonsensical, so I have maintained the original instance names and hierarchy (and copyright).

Compile with

```
$ verilator -cc err_variable_inlining.sv

%Error: Internal Error: blah.sv:16:27: ../V3DfgPasses.cpp:308: Variable inlining should make this impossible
   16 |     input  logic [64-1:0] timer1_timeout_period, 
```

```
// SPDX-License-Identifier: Apache-2.0
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
// http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

module wdt (
    input  logic [64-1:0] timer1_timeout_period, // << --
    output logic t1_timeout
);
    logic [64-1:0] timer1_count;
    assign t1_timeout = (timer1_count == timer1_timeout_period);
endmodule

module caliptra_top_tb_services( input clk );
    logic  set_wdt_timer1_period;
    logic  reset_wdt_timer_period;

    always @(negedge clk) begin
        if(set_wdt_timer1_period) begin
            force caliptra_top_dut.soc_ifc_top1.timer1_timeout_period = 64'hFFFFFFFF_FFFFFFFF;
        end
        else if(reset_wdt_timer_period) begin
            release caliptra_top_dut.soc_ifc_top1.timer1_timeout_period;
        end
    end
endmodule

module soc_ifc_top();
    logic [1:0][31:0] timer1_timeout_period;
    logic timer1_en, t1_timeout;
    wdt i_wdt (
        .timer1_timeout_period(timer1_timeout_period),
        .t1_timeout(t1_timeout)
    );
endmodule

module caliptra_top();
    soc_ifc_top soc_ifc_top1();
endmodule

module caliptra_top_tb( input core_clk );
    caliptra_top_tb_services tb_services_i( .clk(core_clk) );
    caliptra_top caliptra_top_dut         ();
endmodule
```

KlausKoder, 2024-07-12:
I have no idea what I did there... but it was unintentional.

gezalore, 2024-07-13:
Now fixed on master.

gezalore, 2024-07-13:
Also thanks for the reduced test case, that is very helpful.

PR #5259 (2024-07-13): DFG: Allow inlining of variabels driven from forced vars
Not sure why this was disabled before, but it seems legal to me to change

'forced A' -> 'B' -> 'C'

into

'forced A' -> 'B',
'forced A' -> 'C'

Fixes #5249
commit 25f5db4b (2024-07-13): DFG: Allow inlining of variabels driven from forced vars (#5259)

Not sure why this was disabled before, but it seems legal to me to
change

'forced A' -> 'B' -> 'C'

into

'forced A' -> 'B',
'forced A' -> 'C'

Fixes #5249
closed 2024-07-13 by gezalore