verilator/verilator #5253: Add reference to Surfer waveform viewer to the Verilator FAQ

LucasKl, 2024-07-11:
Currently, the Verilator FAQ says that only GTKWave supports the FST format.
I propose to add a reference to the [Surfer](https://gitlab.com/surfer-project/surfer) waveform viewer as well.
Surfer is a relatively young but feature-rich waveform viewer with support for vcd, fst, and ghw files.
Surfer runs natively on Windows, macOS, and Linux as well as in the browser and as a VSCode extension.

The related section of the FAQ:
```
How do I view waveforms (aka dumps or traces)?

Verilator creates standard VCD (Value Change Dump) and FST files. VCD files are viewable with the open-source GTKWave (recommended), or Dinotrace (legacy) programs, or any of the many closed-source offerings; FST is supported only by GTKWave.
``` 

My proposal:
```
How do I view waveforms (aka dumps or traces)?

Verilator creates standard VCD (Value Change Dump) and FST files. VCD files are viewable with the open-source GTKWave (recommended), Surfer, or Dinotrace (legacy) programs, or any of the many closed-source offerings; FST is supported only by GTKWave and Surfer.
```

If possible, it would be nice to make the first Surfer reference a link to the project website (https://surfer-project.org/).

Thank you very much for considering mentioning Surfer!

Kreijstal, 2024-07-11:
could there be a way to somehow simulate inside surfer?

closed 2024-07-11 by wsnyder

LucasKl, 2024-07-12:
@Kreijstal An integration between Verilator and Surfer sounds interesting. If you have any ideas, feel free to open an issue in the Surfer repository. :)

LucasKl, 2024-07-12:
Great! Thanks for mentioning Surfer.