verilator/verilator #5261: FST Top module cannot be fully hidden

sifferman, 2024-07-13:
```
Ubuntu 22.04.4 LTS
Verilator 5.027 devel rev v5.026-4-g833df6d9e
```

I found that doing `--main-top-name "-"` (https://github.com/verilator/verilator/pull/4249) does not work properly with `--trace-fst`.

```systemverilog
module tb;
    wire a = 0;
    initial begin
        $dumpfile( "dump.vcd" );
        $dumpvars;
        $finish;
    end
endmodule
```

## Without `--trace-fst`

```shell
verilator tb.sv \
          --binary \
          --main-top-name "-" \
          --trace
```
[image]

## With `--trace-fst`
```shell
verilator tb.sv \
          --binary \
          --main-top-name "-" \
          --trace-fst
```
[image]

Looping in @dwilliamson in case.

I can look into this when I get some time.

wsnyder, 2024-07-14:
Thanks for the good test case and write up, made it easy to resolve, fixed.

closed 2024-07-14 by wsnyder

sifferman, 2024-07-14:
Thanks! The fix works perfectly