verilator/verilator #5262: fatal error: coroutine: No such file or directory

gustad, 2024-07-15:
I just installed Verilator 5.026 on a Rocky Linux 8.10 based system. Whenever I try to run this installation of Verilator using `verilator --binary` I get:

```
In file included from /net/hansa/tools/verilator/v5.026/share/verilator/include/verilated_timing.cpp:23:
/net/hansa/tools/verilator/v5.026/share/verilator/include/verilated_timing.h:51:11: fatal error: coroutine: No such file or directory
 # include <coroutine>
           ^~~~~~~~~~~
compilation terminated.
make: *** [/net/hansa/tools/verilator/v5.026/share/verilator/include/verilated.mk:274: verilated_timing.o] Error 1
```

I guess this might be because I need to enable C++20? Since this Rocky 8.10 based system is using gcc 8.5.0 g++ should be passed -std=c++2a, but is this supposed to be taken care of by configure, or do I need to pass this by setting CXXFLAGS? If yes, should it be done during configure, make and/or verilator invocation?

wsnyder, 2024-07-15:
This is probably already fixed, please try git master version.

gustad, 2024-07-15:
I get the same result when using the latest master (67ea819d8):

```
$ /tmp/delete-me/bin/verilator --version
Verilator 5.027 devel rev v5.026-49-g67ea819d8
```

```
In file included from /tmp/delete-me/share/verilator/include/verilated_timing.cpp:23:
/tmp/delete-me/share/verilator/include/verilated_timing.h:51:11: fatal error: coroutine: No such file or directory
 # include <coroutine>
           ^~~~~~~~~~~
compilation terminated.
make: *** [/tmp/delete-me/share/verilator/include/verilated.mk:274: verilated_timing.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from Vtestbench___024root.h:9,
                 from Vtestbench__Syms.h:18,
                 from Vtestbench__pch.h:23,
                 from Vtestbench.cpp:4,
                 from Vtestbench__ALL.cpp:3:
/tmp/delete-me/share/verilator/include/verilated_timing.h:51:11: fatal error: coroutine: No such file or directory
 # include <coroutine>
           ^~~~~~~~~~~
compilation terminated.
make: *** [/tmp/delete-me/share/verilator/include/verilated.mk:265: Vtestbench__ALL.o] Error 1
```

I have no problems building and running 5.026 on a Gentoo system (gcc 13.2.1). It appears to be related to Rocky 8 (gcc 8.5.0) somehow.

Setting CXXFLAGS to `-std=gnu++2a` or `-std=c++2a` during the invocation of verilator did not help other than observing that it was passed to the verilator compilation stage:
```
make: Entering directory '/tmp/runsim/cnt4/obj_dir'
g++ -Os -std=gnu++2a -I.  -MMD -I/tmp/delete-me/share/verilator/include -I/tmp/delete-me/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT    -c -o verilated.o /tmp/delete-me/share/verilator/include/verilated.cpp
...
```

wsnyder, 2024-07-15:
Did ./configure say that coroutines are supported?  Given your compiler is old enough to want the flag gnu++2a rather than the c++20 standard, this is likely something we won't support.  But if `./configure "CXX=g++ -std=gnu++2a"` works for you then you are all set.

gustad, 2024-07-15:
Unfortunately, it appears that gcc 8.5.0 does not support corutines when using `-std=gnu++2a` (nor '-std=c++2a`).

On my gentoo system:

```
$ git rev-parse --short HEAD
34e37d7bd
$ g++ --version
g++ (Gentoo 13.2.1_p20240210 p14) 13.2.1 20240210
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ ./configure --prefix=/tmp/delete-me
...
$ grep coroutine config.log
configure:6135: checking whether g++ accepts -fcoroutines-ts
configure:6148: g++ -o conftest   -fcoroutines-ts -Werror     conftest.cpp  >&5
g++: error: unrecognized command-line option '-fcoroutines-ts'; did you mean '-fcoroutines'?
configure:6176: g++ -c   -fcoroutines-ts -Werror   conftest.cpp >&5
g++: error: unrecognized command-line option '-fcoroutines-ts'; did you mean '-fcoroutines'?
configure:6215: checking whether g++ accepts -fcoroutines
configure:6228: g++ -o conftest   -fcoroutines -Werror     conftest.cpp  >&5
configure:6256: g++ -c   -fcoroutines -Werror   conftest.cpp >&5
configure:6446: checking whether coroutines are supported by g++
configure:6466: g++ -o conftest   -fcoroutines     conftest.cpp  >&5
CFG_CXXFLAGS_COROUTINES='-fcoroutines'

```

On my Rocky Linux 8.10 system:

```
$ git rev-parse --short HEAD
34e37d7bd
$ g++ --version
g++ (GCC) 8.5.0 20210514 (Red Hat 8.5.0-22)
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ ./configure "CXX=g++ -std=gnu++2a" --prefix=/tmp/delete-me
...
$ grep coroutine config.log
configure:5468: checking whether g++ -std=gnu++2a accepts -fcoroutines-ts
configure:5481: g++ -std=gnu++2a -o conftest   -fcoroutines-ts -Werror     conftest.cpp  >&5
g++: error: unrecognized command line option '-fcoroutines-ts'
configure:5508: g++ -std=gnu++2a -c   -fcoroutines-ts -Werror   conftest.cpp >&5
g++: error: unrecognized command line option '-fcoroutines-ts'
configure:5546: checking whether g++ -std=gnu++2a accepts -fcoroutines
configure:5559: g++ -std=gnu++2a -o conftest   -fcoroutines -Werror     conftest.cpp  >&5
g++: error: unrecognized command line option '-fcoroutines'; did you mean '-fc-prototypes'?
configure:5586: g++ -std=gnu++2a -c   -fcoroutines -Werror   conftest.cpp >&5
g++: error: unrecognized command line option '-fcoroutines'; did you mean '-fc-prototypes'?
configure:5624: checking whether g++ -std=gnu++2a accepts -fcoroutines-ts -Wno-deprecated-experimental-coroutine
configure:5637: g++ -std=gnu++2a -o conftest   -fcoroutines-ts -Wno-deprecated-experimental-coroutine -Werror     conftest.cpp  >&5
g++: error: unrecognized command line option '-fcoroutines-ts'
configure:5664: g++ -std=gnu++2a -c   -fcoroutines-ts -Wno-deprecated-experimental-coroutine -Werror   conftest.cpp >&5
g++: error: unrecognized command line option '-fcoroutines-ts'
configure:5771: checking whether coroutines are supported by g++ -std=gnu++2a
conftest.cpp:25:10: fatal error: coroutine: No such file or directory
 #include <coroutine>
| #define __cpp_impl_coroutine 1
| #include <coroutine>
conftest.cpp:21:10: fatal error: experimental/coroutine: No such file or directory
 #include <experimental/coroutine>
| #include <experimental/coroutine>
```

Seems like I need to install a more recent gcc...

closed 2024-07-16 by wsnyder

jchap2k, 2024-09-10:
This update doesn't seem to work for me. My company uses GCC 9.2.0 and platform architect, (so I have to use the gcc that they use).   As far as I can find, their GCC install doesn't have coroutine.   Is there any way to still use verilator's newer versions without this?  It is quite limiting.    If it were a personal project it wouldn't be an issue to just change the GCC, recompile SystemC, and move right along, but requiring C++20  is very limiting.  Might not be in a few years, but right now it is.

wsnyder, 2024-09-10:
The configure scripts should detect no coroutines and work fine without timing support.  Note we regress this with Ubuntu 20.04 currently, so it at least works in some cases, you'll need to see why it doesn't work in yours.  You should see HAVE_COROUTINES be undef in src/config_package.h.

jchap2k, 2024-09-11:
Very odd.  Will look into it again.

ahmedelz, 2025-03-21:
I ran into this same issue trying to compile v5.034 on RHEL 8. In src/config_package.h `#undef HAVE_COROUTINES` was commented out in my case. I tried to compile with it being commented out but still ran into the error above running with `--binary`. Then I tried to re-compile again with the undef HAVE_COROUTINES line uncommented but still ran into the same error.

```
g++ -Os  -I.  -MMD -I/<redacted>/verilator/v5.034/share/verilator/include -I/<redacted>/verilator/v5.034/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TIMING=1 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-subobject-linkage -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT    -c -o verilated_timing.o /<redacted>/verilator/v5.034/share/verilator/include/verilated_timing.cpp
In file included from /<redacted>/verilator/v5.034/share/verilator/include/verilated_timing.cpp:23:
/<redacted>/verilator/v5.034/share/verilator/include/verilated_timing.h:51:11: fatal error: coroutine: No such file or directory
 # include <coroutine>
           ^~~~~~~~~~~
compilation terminated.
make[1]: *** [/<redacted>/verilator/v5.034/share/verilator/include/verilated.mk:281: verilated_timing.o] Error 1
```

wsnyder, 2025-03-21:
Please attach a log of running "./configure"

ahmedelz, 2025-03-21:
[verilator_configure.txt](https://github.com/user-attachments/files/19397291/verilator_configure.txt)
Please see attached

wsnyder, 2025-03-21:
>compiler g++ --version = g++ (GCC) 8.5.0 20210514 (Red Hat 8.5.0-24)

Your compiler doesn't support coroutines.  You can't use --timing or --binary.  If you want that you need to install a more modern compiler, or use a more modern OS distribution.