verilator/verilator #5265: Unsupported: virtual interface never assigned any actual interface

solomatnikov, 2024-07-15:
```
%Error-UNSUPPORTED: test2.sv:740:12: Unsupported: virtual interface never assigned any actual interface
  740 |    virtual z_driver_if vif;
      |            ^~~~~~~~~~~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=5.027
%Error-UNSUPPORTED: test2.sv:741:25: Unsupported: virtual interface never assigned any actual interface
  741 |    function new(virtual z_driver_if _vif);
      |                         ^~~~~~~~~~~
%Error: Exiting due to 2 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

VCS compiles without errors.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
class z_bfm;  
   virtual z_driver_if vif;
   function new(virtual z_driver_if _vif);
      vif = _vif;
   endfunction  
   virtual function void init;
   endfunction  
endclass
module test_cw;
  (
   input  logic     clk,
   input  logic     reset_l,
   z_if.sender    z,
                    mmio_z
   );
   z_driver_if    mem_z_drv(clk, z),
                    mmio_z_drv(clk, mmio_z);
   z_bfm mem_z_bfm,
           mmio_z_bfm;
   initial begin
      mem_z_bfm  = new(mem_z_drv);
      mmio_z_bfm = new(mmio_z_drv);
      mem_z_bfm.init;
      mmio_z_bfm.init;
   end
endmodule
```

`test_cw` is never used/instantiated.

What 'verilator' command line do we use to run your example?

```
verilator -Wno-LATCH -Wno-STMTDLY -Wno-WIDTH -Wno-UNUSED --cc --exe -sv --assert --trace --trace-params --trace-structs --report-unoptflat --no-timing --timescale 1ps/1ps --x-assign unique -fno-const-bit-op-tree -fno-dfg-peephole -Wno-SYMRSVDWORD --top-module top test2.sv
```

What 'verilator --version' are you using?  Did you try it with the git master version?

```
v5.026-g34e37d7bd
```

What OS and distribution are you using?

```
Linux 5.14.0-427.24.1.el9_4.x86_64
```

sterin, 2024-07-22:
Simpler examples:

```systemverilog
interface I;
endinterface

class C;  
   virtual I vi;
endclass

module top;
endmodule
```

results in

```
%Error-UNSUPPORTED: bug.sv:5:12: Unsupported: virtual interface never assigned any actual interface
    5 |    virtual I vi;
      |            ^
```

and

```systemverilog
interface I;
endinterface

class C;  
   task automatic t(virtual I vif);
   endtask
endclass

module top;
endmodule
```

results in 

```
%Error-UNSUPPORTED: bug.sv:5:29: Unsupported: virtual interface never assigned any actual interface
    5 |    task automatic t(virtual I vif);
      |                             ^
```

Both with the following arguments:

```
verilator --timing --trace-fst --top-module top --binary bug.sv
```

wsnyder, 2024-07-22:
As it says this isn't currently supported.  I presume your request is to add support for this.

@kozdra is there a reason we can't just ignore unused virtual interfaces like this?

kozdra, 2024-07-23:
Use of an actual interface as a value of virtual interface dtype is required in V3LinkDot, because unused interfaces are marked dead and not resolved correctly afterwards.
We could try to mark them as not dead, but I am not sure if just doing it would work without plugging them as used cells anywhere. If I remember correctly, further stages might assume that interfaces are either dead or used by actual cells, in that case this feature would require modifying their checks.

wsnyder, 2025-04-26:
BTW if #2227 is fixed, Verilator will detect the unused class and this error will go away in the cases mentioned.

solomatnikov, 2025-05-05:
More examples from https://github.com/pulp-platform/common_cells

```
%Error-UNSUPPORTED: common_cells/test/stream_test.sv:26:13: Unsupported: virtual interface never assigned any actual interface
   26 |     virtual STREAM_DV #(
      |             ^~~~~~~~~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=5.037
%Error-UNSUPPORTED: common_cells/test/stream_test.sv:31:15: Unsupported: virtual interface never assigned any actual interface
   31 |       virtual STREAM_DV #(
      |               ^~~~~~~~~
```

solomatnikov, 2025-06-26:
This is a **real showstopper** for many projects, including open source projects without UVM!