verilator/verilator #5272: Bad C++ generated with interface array element being assigned to a virtual interface

sterin, 2024-07-17:
This issue is probably a duplicate of #5270. However, it does not produce an internal error. Instead, it genertes C++ code that does not compile. 

At this point any attempt to remove anything results in an internal error rather than bad C++ code. 

```systemverilog
interface I;
endinterface

class C;
  virtual I i;
endclass

module M(I i0);
endmodule

module top;

  I arr[2]();

  M m0(arr[0]);
  M m1(arr[1]);

  C c;
  initial begin
    C c = new();
    c.i = arr[0];
  end

endmodule
```

When running this code under Verilator, the C++ then produces the following errors:

```
make: Entering directory '/home/user/workspace/systemverilog/obj_dir'
ccache g++ -Os  -I.  -MMD -I/home/user/.local/share/verilator/include -I/home/user/.local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -c -o verilated.o /home/user/.local/share/verilator/include/verilated.cpp
ccache g++ -Os  -I.  -MMD -I/home/user/.local/share/verilator/include -I/home/user/.local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -c -o verilated_vcd_c.o /home/user/.local/share/verilator/include/verilated_vcd_c.cpp
ccache g++ -Os  -I.  -MMD -I/home/user/.local/share/verilator/include -I/home/user/.local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -c -o verilated_threads.o /home/user/.local/share/verilator/include/verilated_threads.cpp
/usr/bin/python3 /home/user/.local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp Vtop___024root__DepSet_h84412442__0.cpp Vtop___024root__DepSet_heccd7ead__0.cpp Vtop_I__DepSet_h2683e081__0.cpp Vtop___024unit__03a__03aC__Vclpkg__DepSet_h74ff9d40__0.cpp Vtop___024unit__03a__03aC__Vclpkg__DepSet_hbc32e5ab__0.cpp Vtop__main.cpp Vtop__Trace__0.cpp Vtop___024root__Slow.cpp Vtop___024root__DepSet_h69cd1b79__0__Slow.cpp Vtop___024root__DepSet_heccd7ead__0__Slow.cpp Vtop___024unit__Slow.cpp Vtop___024unit__DepSet_hff17caec__0__Slow.cpp Vtop_I__Slow.cpp Vtop_I__DepSet_h2683e081__0__Slow.cpp Vtop___024unit__03a__03aC__Vclpkg__Slow.cpp Vtop___024unit__03a__03aC__Vclpkg__DepSet_hbc32e5ab__0__Slow.cpp Vtop__Syms.cpp Vtop__Trace__0__Slow.cpp Vtop__TraceDecls__0__Slow.cpp > Vtop__ALL.cpp
ccache g++ -Os  -I.  -MMD -I/home/user/.local/share/verilator/include -I/home/user/.local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT   -c -o Vtop__ALL.o Vtop__ALL.cpp
In file included from Vtop__ALL.cpp:12:
Vtop___024root__DepSet_h69cd1b79__0__Slow.cpp: In function ‘void Vtop___024root___eval_initial__TOP(Vtop___024root*)’:
Vtop___024root__DepSet_h69cd1b79__0__Slow.cpp:19:20: error: cannot convert ‘Vtop___024root’ to ‘Vtop_I*’ in assignment
   19 |         = vlSelf[0U];
      |           ~~~~~~~~~^
      |                    |
      |                    Vtop___024root
make: *** [/home/user/.local/share/verilator/include/verilated.mk:265: Vtop__ALL.o] Error 1
make: Leaving directory '/home/user/workspace/systemverilog/obj_dir'
%Error: make -C obj_dir -f Vtop.mk -j 1 exited with 2

```

> What 'verilator' command line do we use to run your example?

`verilator --timing --trace --top-module top --binary bug.sv`

> What 'verilator --version' are you using?  Did you try it with the git master version?

Tried on Verilator 5.026 and current master branch.

> What OS and distribution are you using?

Linux under WSL.

> May we assist you in trying to fix this in Verilator yourself?

If this look like someone who is a Verilator beginner could reasonably handle.

wsnyder, 2024-07-24:
Thanks for the report.  From this and other virtual interface issues these may take some time to get to resolved, so pull requests to fix these welcome.

wsnyder, 2025-09-09:
Currently passes on master, fixed some time in past.