verilator/verilator #5274: Linting exits on UNSUPPORTED feature

RobyB, 2024-07-17:
I am trying to lint some code that has one included file that contains some unsupported SystemVerilog constructs.

When i lint the file verilator exits with:
%Error-UNSUPPORTED: FileName.sv Unsupported: highz strength
%Error: Verilator internal fault, sorry.

The unsupported syntax is
  607 |     buf (highz0, pull1) pu (pad, (pe & ps));
      |          ^~~~~~

It is fine for me to know that Verilator does not support Highz strenth
but i would like to simply waive this Error and go on with the linting of the remaining code.

Apparently there is no way to skip this linting error, or i didn't find it.
I tried to add in code pragmas without success
Adding Wno-{} does not work
Adding a config file containing
`verilator_config
lint_off -file "File.sv"

does not exclude the file (with full or relative path) from linting [probably because it is an included file]

So i am blocked and can not use the tool at all.

There has to be something i am missing here.

I would expect file exclusion to work (even if not the preferred solution) or to be able to exclude this check with something like
-Wno-{UNSUPPORTED}
Or just printing out this failure but not flagging it as an Error, but only as a Warning.

wsnyder, 2024-07-17:
The intent is you may use -Wno-UNSUPPORTED, however Verilator needs to try to recover from that and there's a ton of possibilities that it might not get right and crash later.

>%Error: Verilator internal fault, sorry.

Is saying it couldn't recover, which is why disabling UNSUPPORTED didn't work.

If you want to look into why your example crashes, please submit a new issue with a test case, and ideally look into a pull to make the UNSUPPORTED not crash.  Usually these are relatively easy to fix (some null pointer mishandled for example).

Otherwise use an `ifndef VERILATOR around cell primitive use.

Closing as either can suppress, or need separate issue to prevent crash.

closed 2024-07-17 by wsnyder

RobyB, 2024-07-17:
Thx, for the fast answer. 
will try your suggestions and reply back.