verilator/verilator #5275: Semantics of Verilog logical shift (<<) operator

chili-chips-ba, 2024-07-17:
We are looking for your expert opinion on an interesting question that pertains to Verilog language and simulation results from @gtaylormb

```
module calc_envelope_shift
    import opl3_pkg::*;
(
    input wire clk,
    input wire [REG_BLOCK_WIDTH-1:0] block
);
    logic [REG_BLOCK_WIDTH:0] block_shifted;

    always_comb
        block_shifted = block << 1;
```

REG_BLOCK_WIDTH=3, **block=3'b111**, **block_shifted gets 4'b1110**.

[image]

> There are no synthesis warnings and no other tools complain because the original syntax is correct. This type of shift is all over the code and proven in both Xilinx and Altera FPGAs.

The counter-claim is that the **block_shifted** should have gotten **4'b0110**, and that Verilator Lint warning is correct:

```
%Warning-WIDTHEXPAND: modules/operator/src/calc_envelope_shift.sv:119:31: 
Operator SHIFTL expects 4 bits on the LHS, but LHS's VARREF 'block' generates 3 bits.
  : ... note: In instance 'opl3.channels.control_operators.operator.envelope_generator.calc_envelope_shift'
  119 |         block_shifted = block << 1;
```
As the discussion on this topic is still developing, you can check this [link](https://github.com/chili-chips-ba/openCologne/issues/16#issuecomment-2232563143) for additional detail. The same question was also posed to [AccelleraForum](https://forums.accellera.org/topic/7906-semantics-of-verilog-logical-shift)

wsnyder, 2024-07-17:
I think IEEE 11.8.3 is clear that the expression width is extended to match the assignment (to 4 bits).  The table referenced in that other bug is only for self-determined expressions which doesn't apply.  I agree the spec is confusing.

Verilator warns because this may be surprising to the coder, which it seems it was ;)

closed 2024-07-17 by wsnyder

gtaylormb, 2024-07-17:
> Verilator warns because this may be surprising to the coder, which it seems it was ;)

I would argue the warning is more confusing, because it appears something is wrong when there is not. I would guess this probably results in people either making their code unnecessarily verbose to satisfy the warning thinking they're actually fixing something potentially introducing an actual error, or turning off the warning completely (which is what I have done).

An additional point is that for actual loss of precision, there is no warning at all:

```
module calc_envelope_shift
    import opl3_pkg::*;
(
    input wire clk,
    input wire [REG_BLOCK_WIDTH-1:0] block
);
    logic [REG_BLOCK_WIDTH-1:0] block_shifted;

    always_comb
        block_shifted = block << 1;
```

wsnyder, 2024-07-17:
All WIDTH warnings are something that is legal code but possibly not doing what's expected, based on 15 years of what seems to be common coding styles or not.  The case you later indicate most designers will get what they expect, the value moved over one bit into the wider bus - why else would they have made the output one bit wider?

gtaylormb, 2024-07-17:
> The case you later indicate most designers will get what they expect, the value moved over one bit into the wider bus - why else would they have made the output one bit wider?

Exactly. Why is there a warning?