verilator/verilator #5276: Internal Error: Verilator threw signal 9

luizademelo, 2024-07-17:
Hi,
When running "verilator -cc" on the file [test11018.txt](https://github.com/user-attachments/files/16265092/test11018.txt) Verilator crashes with the following message:
```
%Error: Verilator threw signal 9. Suggest trying --debug --gdbbt
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /usr/local/share/verilator/bin/verilator_bin -cc test11018.v
```
When running with "--debug" and "--gdbbt" option, this is the output: 
```
Program terminated with signal SIGKILL, Killed.
The program no longer exists.
No stack.
%Error: export VERILATOR_ROOT=/usr/local/share/verilator
%Error: ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize gdb /usr/local/share/verilator/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run -cc test11018.v --debug --gdbbt" -ex 'set width 0' -ex 'bt' -ex 'quit'
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec setarch --addr-no-randomize gdb /usr/local/share/verilator/bin/verilator_bin_dbg --batch --quiet --return-child-result -ex "run -cc test11018.v --debug --gdbbt" -ex 'set width 0' -ex 'bt' -ex 'quit'
```
I am running the following Verilator version:
Verilator 5.027 devel rev v5.026-29-ge3e7b3f2b

This program was accepted on the Jasper Formal Verification Platform from Cadence Design Systems, so I think this program is correct.

wsnyder, 2024-07-24:
I suspect you are running out of memory? It needs 31 GB on my machine.  How much do you have?

It looks like we're over optimizing this and it's taking a long time. There's probably a lot of tuning we could do to make this Verilate much faster.  FWIW generally we haven't spent a lot of time optimizing Verilation time of gate level(ish) netlists like this.

It makes progress with a stage every few seconds until V3DfgOptimizer at which point after 10 minutes I gave up - @gezalore could you look into what's making Dfg explode please?  When I stopped it, it was deep inside **V3DfgPasses::cse**.

gezalore, 2024-07-25:
On my machine, with current master, running this **without** --debug (to be precise, with these options: `-cc --stats -Wno-fatal`), takes ~90 seconds and ~25.7GB memory.

Running the same adding -fno-dfg, it takes ~97 seconds and ~26.1GB memory.

In both cases the peak memory consumption is introduced in the tristate pass, which is before DFG.

DFG gets rid of a lot of redundancy which does speed up later passes. Without DFG, the same is picked up in V3Gate/V3Const.

With that in mind I will claim DFG does a pretty good job here, but I will see if the speed can be improved in some way.

closed 2026-10-18 by wsnyder