verilator/verilator #5281: V3LinkDot internal error when calling a function on a non-instantiated interface

b-chmiel, 2024-07-18:
# The problem
I get the following internal error when calling a function on a non-instantiated interface. 

```
%Error: Internal Error: t/t_linkdot_bug.v:7:11: ../V3LinkDot.cpp:422: Module/etc never assigned a symbol entry?
-node: IFACE 0x555556cbbc20 <e824#> {e7al}  iface  L3 [LIB] [DEAD] [1ps]
    7 | interface iface;
      |           ^~~~~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

# Test case

t_linkdot_bug.pl
```perl
#!/usr/bin/env perl
if (!$::Driver) { use FindBin; exec("$FindBin::Bin/bootstrap.pl", @ARGV, $0); die; }
# DESCRIPTION: Verilator: Verilog Test driver/expect definition
#
# Copyright 2024 by Wilson Snyder. This program is free software; you
# can redistribute it and/or modify it under the terms of either the GNU
# Lesser General Public License Version 3 or the Perl Artistic License
# Version 2.0.
# SPDX-License-Identifier: LGPL-3.0-only OR Artistic-2.0

scenarios(vlt => 1);

compile(
    );

execute(
    check_finished => 1,
    );

ok(1);
1;
```
t_linkdot_bug.v
```verilog
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by Antmicro.
// SPDX-License-Identifier: CC0-1.0

interface iface;
   function static func;
   endfunction
endinterface

module t;
  initial begin
    iface::func();
    $write("*-* All Finished *-*\n");
    $finish;
   end
endmodule
```

Note: the error does not appear when the function is called on instantiated interface:

```verilog
module t;
  iface iface();
  initial begin
    iface.func();
    $write("*-* All Finished *-*\n");
    $finish;
   end
endmodule
``` 

# Verilator command line

```bash
verilator --exe --main --debug t_linkdot_bug.v
```

# Verilator version

Master at https://github.com/verilator/verilator/commit/095b1ccb6772259082301c672afa2273c60b6688

wsnyder, 2024-07-24:
It's illegal to call "interface::", that only works with classes.  I'll purpose this ticket to add a sane error message.

wsnyder, 2024-11-26:
Fixed with #5629, test added now.

closed 2024-11-26 by wsnyder