verilator/verilator #5286: Cannot find verilator_bin at verilator/bin and getting "sh: line 1: exec: verilator_bin: not found"

dragon540, 2024-07-19:
I am trying to build verilator from the source and following the Quick Install guide given in section 4.1 of the verilator docs. After completing the installation process when I am trying to run the `verilator --version` or` verilator --cc [options] [source_files.v]... [opt_c_files.cpp/c/cc/a/o/so]`, I am getting the following error.

```
sh: line 1: exec: verilator_bin: not found
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec verilator_bin --version
```
Strangely I don't get any error when I run `verilator --help`, and it outputs the verilator manual to the terminal correctly.

I have also tried to unset the VERILATOR_ROOT variable but that doesn't seem to change anything. I think the main problem is not that `verilator_bin` cannot be located properly but that it doesn't even exist under `verilator/bin` directory.

Here are the list files and executables present under `verilator/bin` directory.

````
shobhit@localhost:~/verilator/bin> ls
redirect   verilator_ccache_report  verilator_difftree  verilator_includer
verilator  verilator_coverage       verilator_gantt     verilator_profcfunc
````

How to resolve this?

wsnyder, 2024-07-19:
Did you try starting a new shell?

Can you attach a complete log of every step?

dragon540, 2024-07-19:
> Did you try starting a new shell?
> 
> Can you attach a complete log of every step?

Yes, I tried running it from a new shell but still getting same issue.

Unfortunately I can't attach log of the steps I ran during installation process since I closed that session and can't find it anymore, but I followed section 4.1 from the verilator docs as it is.

wsnyder, 2024-07-23:
I still suspect something in configure or install, to help debug please redo the process from scratch and attach the log, also at the end run 'verilator --debug' and 'printenv' thanks.

wsnyder, 2024-08-24:
I'm assuming you got this resolved.

closed 2024-08-24 by wsnyder

Wren6991, 2024-12-09:
If anyone else finds this from google: I had this exact issue because the Linux CMake install (Ubuntu 24.04 LTS) was building the local verilator binary as `./src/verilator`, installing it with the same name, and then overwriting it with the perl scripts from the `bin/` directory (which include the `verilator` front-end script).

I was able to resolve by manually installing the verilator binary with the name `verilator_bin`. The root cause is probably somewhere in the install rules in `src/CMakeLists.txt`.

wsnyder, 2024-12-09:
Glad you were able to debug. Note the cmake files for Verilator itself (vs Verilated models) are not supported for Linux, they print a warning. Use 'make install'.