verilator/verilator #5288: Consider revamping CDC Linting feature

chili-chips-ba, 2024-07-20:
In the spirit of Verilator tagline `"Open-source SystemVerilog simulator and lint system"`, and even more in the light of acute and urgent dev community needs, may we propose to seriously consider dusting off the CDC Linting capability.

The CDC Linting feature add-on was also proposed to Yosys [dev](https://github.com/YosysHQ/yosys/issues/4493#issue-2411502101) community.

wsnyder, 2024-07-20:
As noted Verilator used to have limited CDC functions. This was rarely (if ever) used outside one organization.

I'm open to this being added back, but it will need someone(s) willing to sign up to dust off the code, improve it for what's needed, and support it long term.

chili-chips-ba, 2024-07-20:
Any insights into why interest level in CDC was so low in the past? We love Verilator linting, and Verilator is the essential part of our QA flow. 

With the vast pool of open-source IP libraries (many of questionable quality), and more to come, such as [this](https://nlnet.nl/project/Gateware-Repository), the importance of formalized RTL quality checking has never been greater.

We are spreading the word about Verilator team looking for help with this honorable task...

wsnyder, 2024-07-20:
Most are using closed-source CDC, and the remainder perhaps don't even consider CDC ;)

markusdd, 2024-07-26:
I just can say that we actually used it quite excessively because it was useful in cleaning up a lot of stuff before 'shooting the big guns'.
For SoC sign-off a commercial tool is usually required but what verilator offered was more than good enough to already catch 80+% of potential issues.

chili-chips-ba, 2024-07-26:
@markusdd, it sounds like it was not too long ago that Verilator still had the CDC Linting capability. If so, it would probably not be too much effort to resurrect it. 

In the meantime, we are kicking [cdc_snitch](https://github.com/chili-chips-ba/openCologne/issues/26#issue-2425828365) tires. @ldoolitt, cdc_snitch author, is looking for testers and first-hand community feedback on his cool little tool...

chili-chips-ba, 2024-11-23:
Hi Verilator experts. 
 Any update on this?
 
With our [Wireguard-FPGA](https://github.com/chili-chips-ba/wireguard-fpga/issues/7) project ramping up, we'd like to revisit the question of tool to use for CDC linting.

wsnyder, 2024-11-23:
Still as above: "I'm open to this being added back, but it will need someone(s) willing to sign up to dust off the code, improve it for what's needed, and support it long term.

chili-chips-ba, 2024-11-23:
Are you actively looking / recruiting for a willing developer, or passively waiting for someone to see this issue and sign up?

wsnyder, 2024-11-23:
Waiting for someone to sign up, might you?  (Too many projects to actively recruit them all!)

chili-chips-ba, 2024-11-23:
... we are System Verilog, RTL and bare-metal Embedded folks. Hosted C++ for complex EDA tools is way above our head 🥲.

Have you reached out to @edmundhumenberger from SymbioticEDA?!

wsnyder, 2024-11-23:
No, feel free to recruit ;)

wsnyder, 2025-02-25:
Adding CDC is not planned, as described we will need someone to contribute and agree to support this long term, until then no need to keep this open.

closed 2025-02-25 by wsnyder