verilator/verilator #5289: C++ compilation error when calling a class member accepting a virtual interface with a wrong interface type

sterin, 2024-07-21:
Calling an class member accepting a virtual interface with the wrong interface type, results in C++ compliation error.

Here, an interface of type `I1` is provided as a parameter to a function that accepts type `I2`:

```systemverilog
interface I1;
endinterface

interface I2;
endinterface

class C;
  function new(virtual I2 s);
  endfunction
endclass

module test;
  I2 i2();
  I1 i1();
  initial begin
    C c = new(i1);
  end
endmodule
```

With the following error:

```
In file included from /tmp/verilator/include/verilated.h:1004,
                 from Vtest__pch.h:21,
                 from Vtest.cpp:4,
                 from Vtest__ALL.cpp:3:
/tmp/verilator/include/verilated_types.h: In instantiation of ‘VlClassRef<T_Class>::VlClassRef(VlDeleter&, T_Args&& ...) [with T_Args = {Vtest__Syms* const __restrict__&, Vtest_I1*}; T_Class = Vtest___024unit__03a__03aC]’:
Vtest___024root__DepSet_h1529598b__0__Slow.cpp:18:38:   required from here
/tmp/verilator/include/verilated_types.h:1831:18: error: no matching function for call to ‘Vtest___024unit__03a__03aC::Vtest___024unit__03a__03aC(Vtest__Syms* const __restrict__&, Vtest_I1*)’
 1831 |         : m_objp{new T_Class(std::forward<T_Args>(args)...)} {
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Vtest__ALL.cpp:8:
Vtest___024unit__03a__03aC__Vclpkg__DepSet_hbe81a92c__0.cpp:9:1: note: candidate: ‘Vtest___024unit__03a__03aC::Vtest___024unit__03a__03aC(Vtest__Syms*, Vtest_I2*)’
    9 | Vtest___024unit__03a__03aC::Vtest___024unit__03a__03aC(Vtest__Syms* __restrict vlSymsp, Vtest_I2* s) {
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~
Vtest___024unit__03a__03aC__Vclpkg__DepSet_hbe81a92c__0.cpp:9:99: note:   no known conversion for argument 2 from ‘Vtest_I1*’ to ‘Vtest_I2*’
    9 | Vtest___024unit__03a__03aC::Vtest___024unit__03a__03aC(Vtest__Syms* __restrict vlSymsp, Vtest_I2* s) {
      |                                                                                         ~~~~~~~~~~^
In file included from Vtest__Syms.h:21,
                 from Vtest__pch.h:23,
                 from Vtest.cpp:4,
                 from Vtest__ALL.cpp:3:
Vtest___024unit__03a__03aC__Vclpkg.h:31:7: note: candidate: ‘Vtest___024unit__03a__03aC::Vtest___024unit__03a__03aC(const Vtest___024unit__03a__03aC&)’
   31 | class Vtest___024unit__03a__03aC : public VlClass {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~
Vtest___024unit__03a__03aC__Vclpkg.h:31:7: note:   candidate expects 1 argument, 2 provided
In file included from /tmp/verilator/include/verilated.h:1004,
                 from Vtest__pch.h:21,
                 from Vtest.cpp:4,
                 from Vtest__ALL.cpp:3:
/tmp/verilator/include/verilated_types.h:1831:11: error: cannot convert ‘<brace-enclosed initializer list>’ to ‘Vtest___024unit__03a__03aC*’ in initialization
 1831 |         : m_objp{new T_Class(std::forward<T_Args>(args)...)} {
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

The same error occurrs with tasks, functions, static tasks, and static functions. 

However,  when the interface `I2` is never instantiated, the correct error message is reported:

```
%Error-UNSUPPORTED: bug.sv:8:24: Unsupported: virtual interface never assigned any actual interface
    8 |   function new(virtual I2 s);
      |                        ^~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=5.027
%Error: Exiting due to 1 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

> What 'verilator' command line do we use to run your example?

`verilator --timing --trace --top-module test--binary bug.sv`

> What 'verilator --version' are you using?  Did you try it with the git master version?

Tried on Verilator 5.026 and current git master.

> What OS and distribution are you using?

Ubuntu 22.04 under WSL.

> May we assist you in trying to fix this in Verilator yourself?

If this look like someone who is a Verilator beginner could reasonably handle.

wsnyder, 2024-07-24:
Thanks for the report.  From this and other virtual interface issues these may take some time to get to resolved, so pull requests to fix these welcome.

ArturBieniek4, 2026-09-02:
https://github.com/verilator/verilator/pull/8180 added proper rejection of this illegal shape.

closed 2026-09-02 by ArturBieniek4