verilator/verilator #5295: Segmentation fault on Ubuntu 20.04

jordancarlin, 2024-07-22:
This is a very bizarre bug that shows up intermittently. I'll do my best to explain what's going on here but feel free to clarify as needed.

When running the test suite for https://github.com/openhwgroup/cvw using Verilator, I get failures on certain distros with certain tests. On Ubuntu 22.04/24.04 and Rocky 9.3, Verilator runs without issues. On Ubuntu 20.04 and Rocky 8.9, I get an error message saying `Segmentation fault (core dumped)`. This does not happen with all of our tests, and there does not seem to be a hard and fast rule regarding which ones it does happen for, but I've noticed the following trends. Failures are more likely with larger configurations (though it has happened once or twice with a smaller configuration) and failures are more common the more tests have been run or are being run simultaneously. For example, with a new system, the first few tests work fine, and then by the 5th or 6th they usually start to fail. Then there are usually like 10 failures in a row before getting a few more random successes. I have to remove the compiled files and rerun Verilator on a clean repo to make it work for a few tests again.

For the cases that do fail, the compilation finishes (supposedly successfully), but when running the compiled file (Vtestbench) it immediately hits the segmentation fault. By adding `$display` statements, I have discovered that it is failing even before the first `initial begin` in the testbench.

I have installed and am using gcc-10 since the existing gcc-9 in Ubuntu 20.04 seems to be too old for Verilator.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Due to the intermittent nature of the error, I have been unable to produce a simplified version for debugging, but I created a docker container with all of the tools installed. Download and run using the following commands:
```
docker pull ghcr.io/jordancarlin/cvw/ubuntu20cvw:latest
docker run -ti ubuntu20cvw:latest                       # add any extra volumes that you need for debugging
```
Inside the docker container, compile the tests and then either run all of the tests at once (most likely to give failures) or one test at a time using the commands below. These commands these tests run are all in the attached [Makefile](https://github.com/user-attachments/files/16327801/Makefile.txt) (which can also be seen at `home/wally/cvw/sim/verilator` in the docker container).

```
make riscof                               # compile all tests
regression-wally                          # run all tests
wsim --sim verilator rv64gc arch64zfh     # run one test using script
```

What 'verilator --version' are you using?  Did you try it with the git master version?
Primarily tested on the git master version. Also tried on 5.026

What OS and distribution are you using?
The issue is happening on Ubuntu 20.04 and Rocky 8.9. The same setup on Ubuntu 22.04/24.04 and Rocky 9 works fine.

May we assist you in trying to fix this in Verilator yourself?
I'm not familiar with the inner workings of Verilator, but I'm willing to give it a go if you think it would be easy for a beginner.

wsnyder, 2024-07-22:
Unfortunately my time is fairly limited I'm unlikely to be able to find time to look into debugging complex containerized examples like this soon.

Can you please try --runtime-debug, this adds C++ assertions and often finds issues that are otherwise unpredictable?

jordancarlin, 2024-07-23:
@wsnyder No worries regarding the limited time. Thanks for the help. When running the following verilator command (with --runtime-debug), I get a bunch of warnings about `variable tracking size limit exceeded with '-fvar-tracking-assignments'` and the compilation hangs. It doesn't fail or give any error messages, but it has been on the same g++ step for several hours without making any progress. Not sure where to go from here. 

COMMAND
```
verilator \
--Mdir wkdir/rv64gc_arch64zfh -o Vtestbench \
--binary --trace \
--runtime-debug -DVERILATOR=1 --no-trace-top --stats \
--top-module testbench  --relative-includes \
"-I/home/wally/cvw/config/shared" "-I/home/wally/cvw/config/rv64gc" "-I/home/wally/cvw/config/deriv/rv64gc" \
/home/wally/cvw/sim/verilator/wrapper.c \
 \
/home/wally/cvw/src/cvw.sv /home/wally/cvw/testbench/testbench.sv /home/wally/cvw/testbench/common/*.sv /home/wally/cvw/src/*/*.sv /home/wally/cvw/src/*/*/*.sv
```

OUTPUT
```
make[1]: Entering directory '/home/wally/cvw/sim/verilator/wkdir/rv64gc_arch64zfh'
ccache g++  -I.  -MMD -I/opt/riscv/share/verilator/include -I/opt/riscv/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -fsanitize=address,undefined -D_GLIBCXX_DEBUG -DVL_DEBUG=1 -DVL_TIME_CONTEXT   -fcoroutines -Os -c -o wrapper.o /home/wally/cvw/sim/verilator/wrapper.c
ccache g++ -Os  -I.  -MMD -I/opt/riscv/share/verilator/include -I/opt/riscv/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -fsanitize=address,undefined -D_GLIBCXX_DEBUG -DVL_DEBUG=1 -DVL_TIME_CONTEXT   -fcoroutines -c -o verilated.o /opt/riscv/share/verilator/include/verilated.cpp
ccache g++ -Os  -I.  -MMD -I/opt/riscv/share/verilator/include -I/opt/riscv/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -fsanitize=address,undefined -D_GLIBCXX_DEBUG -DVL_DEBUG=1 -DVL_TIME_CONTEXT   -fcoroutines -c -o verilated_dpi.o /opt/riscv/share/verilator/include/verilated_dpi.cpp
ccache g++ -Os  -I.  -MMD -I/opt/riscv/share/verilator/include -I/opt/riscv/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -fsanitize=address,undefined -D_GLIBCXX_DEBUG -DVL_DEBUG=1 -DVL_TIME_CONTEXT   -fcoroutines -c -o verilated_vcd_c.o /opt/riscv/share/verilator/include/verilated_vcd_c.cpp
ccache g++ -Os  -I.  -MMD -I/opt/riscv/share/verilator/include -I/opt/riscv/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -fsanitize=address,undefined -D_GLIBCXX_DEBUG -DVL_DEBUG=1 -DVL_TIME_CONTEXT   -fcoroutines -c -o verilated_timing.o /opt/riscv/share/verilator/include/verilated_timing.cpp
ccache g++ -Os  -I.  -MMD -I/opt/riscv/share/verilator/include -I/opt/riscv/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -fsanitize=address,undefined -D_GLIBCXX_DEBUG -DVL_DEBUG=1 -DVL_TIME_CONTEXT   -fcoroutines -c -o verilated_threads.o /opt/riscv/share/verilator/include/verilated_threads.cpp
ccache g++ -Os  -I.  -MMD -I/opt/riscv/share/verilator/include -I/opt/riscv/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -fsanitize=address,undefined -D_GLIBCXX_DEBUG -DVL_DEBUG=1 -DVL_TIME_CONTEXT   -fcoroutines -x c++-header Vtestbench__pch.h -o Vtestbench__pch.h.fast.gch
ccache g++ -Os  -I.  -MMD -I/opt/riscv/share/verilator/include -I/opt/riscv/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable  [... cut ...]
ccache g++ -Os  -I.  -MMD -I/opt/riscv/share/verilator/include -I/opt/riscv/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -fsanitize=address,undefined -D_GLIBCXX_DEBUG -DVL_DEBUG=1 -DVL_TIME_CONTEXT   -fcoroutines -include Vtestbench__pch.h.fast -c -o Vtestbench___024root__DepSet_hfc24d085__5.o Vtestbench___024root__DepSet_hfc24d085__5.cpp
```

jordancarlin, 2024-07-23:
Interestingly, the exact same issue happens (both the warning message and the hang) when running Verilator with the `--runtime-debug` flag on Ubuntu 24.04, which works to simulate the tests without the extra flag.

wsnyder, 2024-07-23:
If the compiler hangs that's a compiler bug unrelated to Verilator, up to you if you want to report it to the compiler folks.

To get around this use the separate flags that are documented that runtime-debug sets. See [--runtime-debug](https://verilator.org/guide/latest/exe_verilator.html#cmdoption-runtime-debug)  

Almost certainly the issue is the -fsanitize one, so use all of the other flags instead `--decorations node -CFLAGS -ggdb  -LDFLAGS -ggdb
 -CFLAGS -D_GLIBCXX_DEBUG
 -CFLAGS -DVL_DEBUG=1`

jordancarlin, 2024-07-24:
Running with `--decorations node -CFLAGS -ggdb  -LDFLAGS -ggdb -CFLAGS -D_GLIBCXX_DEBUG -CFLAGS -DVL_DEBUG=1` causes all of the tests to pass (though it obviously runs way slower).

jordancarlin, 2024-07-24:
After testing all of the flags, it seems that as soon as `-CFLAGS -ggdb` is included, all of the tests pass. Not sure what to make of that.

jordancarlin, 2024-07-24:
Not sure how useful the gdb output is without `-CFLAGS -ggdb`, but here is the output for one of the failing tests when compiled with `--decorations node -LDFLAGS -ggdb -CFLAGS -D_GLIBCXX_DEBUG -CFLAGS -DVL_DEBUG=1`

```
Reading symbols from Vtestbench...
(No debugging symbols found in Vtestbench)
(gdb) run +TEST=arch32i
Starting program: /home/wally/cvw/sim/verilator/wkdir/rv32gc_arch32i/Vtestbench +TEST=arch32i
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x155546963700 (LWP 321)]
[New Thread 0x155546762700 (LWP 322)]
[New Thread 0x155546561700 (LWP 323)]
[New Thread 0x155546360700 (LWP 324)]
[New Thread 0x15554615f700 (LWP 325)]
[New Thread 0x155545f5e700 (LWP 326)]
[New Thread 0x155545d5d700 (LWP 327)]
[New Thread 0x155545b5c700 (LWP 328)]
[New Thread 0x15554595b700 (LWP 329)]
[New Thread 0x15554575a700 (LWP 330)]
[New Thread 0x155545559700 (LWP 331)]
[New Thread 0x155545358700 (LWP 332)]
[New Thread 0x155545157700 (LWP 333)]
[New Thread 0x155544f56700 (LWP 334)]
[New Thread 0x155544d55700 (LWP 335)]
[New Thread 0x155544b54700 (LWP 336)]
[New Thread 0x155544953700 (LWP 337)]
[New Thread 0x155544752700 (LWP 338)]
[New Thread 0x155544551700 (LWP 339)]
[New Thread 0x155544350700 (LWP 340)]
[New Thread 0x15554414f700 (LWP 341)]
[New Thread 0x155543f4e700 (LWP 342)]
[New Thread 0x155543d4d700 (LWP 343)]
[New Thread 0x155543b4c700 (LWP 344)]
[New Thread 0x15554394b700 (LWP 345)]
[New Thread 0x15554374a700 (LWP 346)]
[New Thread 0x155543549700 (LWP 347)]
[New Thread 0x155543348700 (LWP 348)]
[New Thread 0x155543147700 (LWP 349)]
[New Thread 0x155542f46700 (LWP 350)]
[New Thread 0x155542d45700 (LWP 351)]

Thread 1 "Vtestbench" received signal SIGSEGV, Segmentation fault.
0x0000555555721654 in Vtestbench___024unit___eval_static__TOP____024unit(Vtestbench___024unit*) ()
```

jordancarlin, 2024-07-24:
By compiling it only with debug do you mean just `-CFLAGS -ggdb`? If so, that causes the problem to go away. Any combination of flags that includes `-CFLAGS -ggdb` causes the problem to go away.

wsnyder, 2024-07-24:
I meant you could try compiling only that single .o file with debug and no others (and then link).  You'd need to manually do this or edit the makefiles.  Another possibility is to try keeping on optimization with debug (e.g. "-Os -ggdb").

wsnyder, 2024-08-24:
I'm assuming you got this resolved, otherwise please reopen with an example we can debug - or ideally a suggested fix.

closed 2024-08-24 by wsnyder

jordancarlin, 2024-08-24:
Sorry. I haven’t had much time to look into this lately but hope to revisit it soon. I’ll update here if I’m able to make any progress.