verilator/verilator #5297: How to make non-blocking assignment behave correctly for systemc converting from RTL

yinweijie, 2024-07-23:
I try to send `invalid` signal to RTL through systemc code at 13ns. `invalid_r` is expected to be set to 1 a cycle later, however, `invalid_r` is set to 1 at the same cycle as `invalid`. Here is the full code:

`sc_main.cpp`
```cpp
// -*- SystemC -*-
// DESCRIPTION: Verilator Example: Top level main for invoking SystemC model
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2017 by Wilson Snyder.
// SPDX-License-Identifier: CC0-1.0
//======================================================================

// For std::unique_ptr
#include <memory>

// SystemC global header
#include <sysc/kernel/sc_simcontext.h>
#include <sysc/kernel/sc_time.h>
#include <systemc>

// Include common routines
#include <verilated.h>
#if VM_TRACE
#include <verilated_vcd_sc.h>
#endif

#include <sys/stat.h>  // mkdir

// Include model header, generated from Verilating "top.v"
#include "Vtop.h"

using namespace sc_core;
using namespace sc_dt;

int sc_main(int argc, char* argv[]) {
    Verilated::traceEverOn(true);

    sc_clock clk{"clk", 10, SC_NS, 0.5, 3, SC_NS, true};
    sc_signal<bool> in_valid;

    const std::unique_ptr<Vtop> top{new Vtop{"top"}};

    top->clk(clk);
    top->in_valid(in_valid);

    // You must do one evaluation before enabling waves, in order to allow
    // SystemC to interconnect everything for testing.
    sc_start(SC_ZERO_TIME);

    VerilatedVcdSc* tfp = nullptr;
    tfp = new VerilatedVcdSc;
    top->trace(tfp, 99);  // Trace 99 levels of hierarchy
    Verilated::mkdir("logs");
    tfp->open("logs/vlt_dump.vcd");

    // Simulate until $finish
    while (sc_time_stamp() < sc_time(100, SC_NS)) {
        // sc_start(1, SC_NS);

        if((sc_time_stamp() >= sc_time(13, SC_NS)) && (sc_time_stamp() < sc_time(23, SC_NS))) {
            in_valid = 1;
        } else {
            in_valid = 0;
        }

        // Simulate 1ns
        sc_start(1, SC_NS);
    }

    // Final model cleanup
    top->final();

    // Close trace if opened
    if (tfp) {
        tfp->close();
        tfp = nullptr;
    }

    return 0;
}
```

`top.v`
```systemverilog
module top
  (
   // Declare some signals so we can see how I/O works
   input              clk,
   input              in_valid
   );

   reg in_valid_r;

   // Print some stuff as an example
   initial begin
      in_valid_r = 0;
   end

   always @(posedge clk) begin
      in_valid_r <= in_valid;

      $display("[%0t] in_valid_r=%0d\n", $time, in_valid_r);
   end

endmodule

```

`CMakeLists.txt`
```cmake
cmake_minimum_required(VERSION 3.12)
cmake_policy(SET CMP0074 NEW)

project(cmake_tracing_sc_example CXX)

set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

find_package(verilator HINTS $ENV{VERILATOR_ROOT} ${VERILATOR_ROOT})
if (NOT verilator_FOUND)
  message(FATAL_ERROR "Verilator was not found. Either install it, or set the VERILATOR_ROOT environment variable")
endif()

# SystemC dependencies
set(THREADS_PREFER_PTHREAD_FLAG ON)
find_package(Threads REQUIRED)

# Find SystemC using SystemC's CMake integration
find_package(SystemCLanguage QUIET)

# Create a new executable target that will contain all your sources
add_executable(example ./sc_main.cpp)
target_compile_features(example PUBLIC cxx_std_14)

set_property(
  TARGET example
  PROPERTY CXX_STANDARD ${SystemC_CXX_STANDARD}
  )

# Add the Verilated circuit to the target
verilate(example SYSTEMC COVERAGE TRACE
  INCLUDE_DIRS "."
  VERILATOR_ARGS -x-assign fast
  SOURCES ./top.v)

verilator_link_systemc(example)

```

By running:
```bash
> cmake -S . -B build -D CMAKE_BUILD_TYPE=Debug
> cmake --build build
> ./build/example
```
I get:

[image]

`in_valid_r <= in_valid;` seems not give correct result. `in_valid_r` should delay for a cycle.
How to make it works correctly?

wsnyder, 2024-07-23:
Setting a SystemC signal at a time can race (independent of Verilator). Instead make a SC_MODULE to connect to the signal, and that module should have a posedge or negedge SystemC process sensitive to the clock and change the sc_signal there.   Or, just use C++ without SystemC and ensure to change the inputs where the clock is not changing (or on negedge if you have no negedge flops).

yinweijie, 2024-07-24:
@wsnyder 

Thank you for your suggestion. It works after trying your suggestion.
Here is the revised code:

`sc_main.cpp`
```cpp
// -*- SystemC -*-
// DESCRIPTION: Verilator Example: Top level main for invoking SystemC model
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2017 by Wilson Snyder.
// SPDX-License-Identifier: CC0-1.0
//======================================================================

// For std::unique_ptr
#include <memory>

// SystemC global header
#include <sysc/communication/sc_signal.h>
#include <sysc/kernel/sc_simcontext.h>
#include <sysc/kernel/sc_time.h>
#include <systemc>

// Include common routines
#include <verilated.h>
#if VM_TRACE
#include <verilated_vcd_sc.h>
#endif

#include <sys/stat.h>  // mkdir

// Include model header, generated from Verilating "top.v"
#include "Vtop.h"

using namespace sc_core;
using namespace sc_dt;

// make a SC_MODULE to connect to the signal, and that module should have a 
// posedge or negedge SystemC process sensitive to the clock and change the sc_signal.
SC_MODULE(Core) {
    sc_in<bool> clk;
    sc_out<bool> in_valid;

    SC_CTOR(Core) {
        SC_METHOD(process);
        sensitive << clk.pos();
    }

    void process() {
        if((sc_time_stamp() >= sc_time(13, SC_NS)) && (sc_time_stamp() < sc_time(23, SC_NS))) {
            in_valid.write(1);
        } else {
            in_valid.write(0);
        }
    }
};

int sc_main(int argc, char* argv[]) {
    Verilated::traceEverOn(true);

    sc_clock clk{"clk", 10, SC_NS, 0.5, 3, SC_NS, true};
    sc_signal<bool> in_valid{"in_valid"};

    const std::unique_ptr<Vtop> top{new Vtop{"top"}};
    const std::unique_ptr<Core> core{new Core{"core"}};

    top->clk(clk);
    core->clk(clk);
    top->in_valid(in_valid);
    core->in_valid(in_valid);

    // You must do one evaluation before enabling waves, in order to allow
    // SystemC to interconnect everything for testing.
    sc_start(SC_ZERO_TIME);

    VerilatedVcdSc* tfp = nullptr;
    tfp = new VerilatedVcdSc;
    top->trace(tfp, 99);  // Trace 99 levels of hierarchy
    Verilated::mkdir("logs");
    tfp->open("logs/vlt_dump.vcd");

    sc_start(100, SC_NS);

    // Final model cleanup
    top->final();

    // Close trace if opened
    if (tfp) {
        tfp->close();
        tfp = nullptr;
    }

    return 0;
}
```

closed 2024-07-24 by yinweijie