verilator/verilator #5298: Why all threads are launched?

yinweijie, 2024-07-24:
My computer has 16 cores. When verilated executable is launched, all 16 cpu cores are used. 
I try to use parameters to control thread number as:

```cmake
verilate(tb_alu
  VERILATOR_ARGS 
    -Wno-ENUMVALUE
    --threads 6
  TRACE
  INCLUDE_DIRS "."
  SOURCES ./alu.sv ./dpi_func.cpp)
```

However, still 16 cores are launched. How to solve this problem?

[image]

wsnyder, 2024-07-24:
--threads sets only the model's threads, not how many threads your main() makes as you might need threads for multiple models or other reasons.  Set that in your main() with contextp->threads(6).  However, there's no real harm in having unused threads as they won't use CPU cycles.

closed 2024-07-24 by wsnyder

yinweijie, 2024-07-24:
> --threads sets only the model's threads, not how many threads your main() makes as you might need threads for multiple models or other reasons. Set that in your main() with contextp->threads(6). However, there's no real harm in having unused threads as they won't use CPU cycles.

@wsnyder Thanks! By the way, how to determine the number of threads that should be used? I learned from the verilator official guide that a thread should correspond to a model calling `eval()`. Does that mean that, if I only have one model calling `eval()` I should only make 2 threads(One for main, one for `eval()` calling)?

wsnyder, 2024-07-24:
If you aren't using threads in your code, and using just one Verilated model, and want maximum peak performance, then use threads to match your core count.