verilator/verilator #5299: Any Possible Way to Dump DfgGraph with Logic Value on Nodes in Dfg at a Specific Moment of Simulation

enochliang, 2024-07-24:
Hi, I'm working on a project where I need logic values on each nodes of Dfg at a specific moment of simulation.
I can dump a DfgGraph of a CPU design by using `verilator --cc --dumpi-dfg 9 --flatten {design.sv} --top {design_name}`.
I can dump the value of all signals including wires and regs with VCD file, but I need the logic value on operater node (the output of operater), for example, ADD, AND, SEL... as well (the internal signals of verilog expression). They are not included in the VCD file, so I have been considering to run graph propagation on the Dfg to obtain them (propagate the logic values from wires and regs to all the rest of node on the Dfg).
Before I do this, I want to ask:
**Are there any possible way to dump DfgGraph with logic value on every node in Dfg including operators at a specific moment of simulation with verilator?**
**If not, is that possible for AST?**

wsnyder, 2024-07-24:
There is no current "connection" between the Ast dumps and runtime value tracing.

What are you ultimately trying to accomplish?  E.g. perhaps you instead could to add fake signals to every AST output so the trace includes those values?

markusdd, 2024-07-26:
I have no concrete idea for VCD and the internals regarding verilator, but maybe it is useful to look at how the xml2stems mechanism works in GTKWAVE.

If you dump FST traces, open them in gtkwave and also have generated the stems for the rtl viewer, gtkwave displays the live values in the source code depending on where the cursor is.

I think this is what the question is about?

closed 2024-09-08 by wsnyder