verilator/verilator #5301: Test Mac OS in CI

ktbarrett, 2024-07-25:
This is less of a feature of verilator and more of a feature of this project, but to prevent issues like seen in #5190 it would be useful to include a Mac OS regression in CI.

wsnyder, 2024-07-25:
I'd love a pull to add MacOS regressions. To accept it I'll need a primary and backup person that know MacOS to agree to support any portability issues.

ktbarrett, 2024-07-25:
I'm okay with being the primary for the time being (as long as my work laptop is a Mac). I'll also try to get it up and running at some point.

@barbedo? @gussmith23?

barbedo, 2024-07-25:
@ktbarrett, @gussmith23,

I can't commit to being a maintainer, but you can tag me if a regression happens, and I'll try to take a look (if I have the bandwidth).

wsnyder, 2025-02-25:
Given no maintainer has stepped forward & there's no plan to add these tests otherwise, closing as unplanned.

closed 2025-02-25 by wsnyder