verilator/verilator #5304: Calls to mailbox functions which occur as predicate expressions (inside if or assert) are sometimes not evaluated correctly

matthew-humphrey, 2024-07-25:
My team has several modules that we use in simulation to consume or drive common interfaces on hardware modules (for example the read or write side of a FIFO, ready-valid, etc.). Internally these modules use mailboxes to hold the data that has been consumed or is to be driven from/to an interface. In porting from Questa to Verilator, I discovered that these modules were not working. After some experimentation, I discovered that when mailbox calls were embedded within a predicate expression (like inside an `if` or `assert`), the mailbox behavior was not correct. 

I have created a small repro using your test_regress framework. The sample has two mailboxes, one which is populated by a task, and one whose outputs are consumed via a task. In-between the two mailboxesx is a synchronous block (always_ff) that pulls data from the inputs mailbox and pushes it on the outputs mailbox. 

I have tested this with both the current master branch (5.027 devel rev v5.026-75-g0d43c14b9) and the STABLE tag. 

I am using Ubuntu 22.04.4 LTS running within Windows 11 WSL session. 

I am open to fixing the issue myself, with some assistance.

[bug.zip](https://github.com/user-attachments/files/16383878/bug.zip)

wsnyder, 2024-07-25:
You have a statement with side effects inside an assertion, so you need assertions to execute, so the test needs --assert and passes once that is added.  Your code should instead set a variable always and check that variable inside the assert.

`         assert(0 != mb_outputs.try_put(data));
`

matthew-humphrey, 2024-07-25:
Thanks, although I have also observed an issue with the statement inside an "if".

wsnyder, 2024-07-25:
It's likely there's a bug where the if gets optimized away by mistake as there have been other similar bugs.  If you get a reproducer should be an easy fix.  The workaround would be a temporary variable you then test in the if.

matthew-humphrey, 2024-07-25:
Ok, thanks. I think it's OK to close this bug. I can file another one as-needed.

closed 2024-07-25 by wsnyder