verilator/verilator #5307: Python "SRE module mismatch" as Vtop.mk does not set Python executable and verilator_includer hence defaults to system path

markusdd, 2024-07-26:
This does occur with a wide range of varilator versions all the way to latest (probably since verilator_includer was introduced?).

The issue can be seen in action here (I think in this CI Giothub ubuntu it's a 5.020): 
https://github.com/nbiotcloud/ucdp-amba/actions/runs/10102565305/job/27938379707#step:6:578 

Context: We are using cocotb and cocotb test in the linked repository to run some tests on amba infrastructure.
We try to keep compatibility to Python 3.10, 3.11 and 3.12 .

The CI images are Ubuntu 24, hence their system python is 3.12.

As cocotb-test dispatches the verilator compile, it finally uses the generates Vtop.mk to build the simulator executable as is standard practice. This Vtop.mk then also calls verilated.mk which is part of verilators distribution. It contains hardcoded paths for perl and python.

For Perl and the verilator root itself it overrides the variables properly in Vtop.mk, but it does not seem to do that for Python3.
[image]

This then leads to the outcome that the embedded Makefile rules call python3 with the system python, but when you are in a Python venv, as we need to be to have all the cocotb stuff and other python libs (and the actual python version we wanna test against) you run into the situation that whatever the system python is is being called for verilator_includer, but it tries to "import re" from the active venv, which in our case for 3.10 and 3.11 is a different interpretor version and hence the re library import fails.

a) This probably should be harmonized to what is done with perl and respect the Python version present in the environment, as this is an unexpected failure mode

b) as a workaround for others who stumple upon this, cocotb-test aöllow for passing of make_args int the run call, if you have another approach and scripting it needs to be appended to the make command of Vtop.mk, I added PYTHON3=python3 and then the error is gone. This is a bit 'hacky' in my view, but works. Can be seen in action here where the issue has been workarounded and verilator_includer gets called with just "python3" https://github.com/nbiotcloud/ucdp-amba/actions/runs/10108901538/job/27955721755#step:6:350

Still I think this should be fixed because it's somewhat unexpected behavior and working with Python venv's is extremely common, mostly people probably do not just use system python for dev work.

wsnyder, 2024-07-26:
Makes sense to make python match perl in this regard.  Can you please submit a pull that fixes the issue for you?

markusdd, 2024-07-26:
I can try. 2 questions though: 

1) where is the place that makefile gets generated
2) it says in the comment it takes PERL from $PERL, this variable is empty for me though. Does that mean it just defaults to perl if it is not set?

wsnyder, 2024-07-26:
1. V3EmitMk.cpp, but I'd suggest grep for "PERL" and use that as a template for where changes are needed.
2. Correct, in V3Options::getenvPERL.

markusdd, 2024-07-26:
Thanks. Looks like this is not going to be a plain 1-liner, but I'll look into it.

markusdd, 2024-07-26:
ok, that wasn't too difficult.
ready for review.

PR #5308 (2024-07-26): ensure make flows do not call python3 with absolute system path (like perl)
Fixes #5307

local build and make test is passing, verified the ouput of Vtop.mk, which now also ensures PYTHON3 does not default to a full system path to avoid environment mix ups

now looks like this:
[image]
closed 2024-07-26 by wsnyder