verilator/verilator #5309: 'parameter type' not supported in hierarchical blocks

colluca, 2024-07-27:
This is a feature request to support type parameters in hierarchical blocks. I guess opening this issue could be useful to track the popularity of the request, and as a place to discuss design details.

This came up while investigating the possibility to mark core complices in a [Snitch cluster](https://github.com/pulp-platform/snitch_cluster) as hierarchical blocks, in an attempt to reduce memory requirements below 16GB (what public Github runners provide), and ran into this issue.

The error was observed in Verilator 5.006, but from the source code I believe the issue persists also on master:
https://github.com/verilator/verilator/blob/a2496e041ed66af6dec40d29da68a84d65663145/src/V3Param.cpp#L472-L475

I'd be willing to consider contributing this feature, but I have to weigh the required effort against my availability.

Thank you for looking into this :)

wsnyder, 2024-07-27:
If you'd like to work on this a pull would need to roughly:
0. Add appropriate tests
1. Print out what the type needs to be set to (probably V3EmitV is pretty close but might need some tweaks), and pass that to the lower verilation similar to how non-type parameters work.
2. Parse the type parameter string.  This would probably need to use the real parser as types are complicated.  Would also need to resolve data types, e.g. "typedef struct" similar to how V3LinkDot does - presumably assuming the typedef names are global?  Maybe it is better to have a verilog shell module that sets the parameters appropriately rather than using command line?

b-chmiel, 2024-07-29:
I'd experimented in the past with approaches to viably implement such feature and I think the _type parameter string parsing_ may not be the cleanest approach when it comes to the more complex types. For instance, with each parameter type HierBlock parameter parser need to be extended which duplicates already implemented parsing code.

What I propose is instead to use already existing Verilator SV parser and implement it using `files`. For instance, file such as:

```verilog
module __V_type_parameters;
  typedef struct packed { logic val; } VAL_T;
endmodule
```
May be generated during the first hierarchical phase, during generation of hier args files, where each typedef would represent module type parameter .
This way, during hier sub runs (when `v3Global.opt.hierChild()` is set), it will be parsed alongside with hierarchical submodules code and then during `V3LinkDot` extracted from this helper files.

As a naive approach, those typedefs may be duped into `V3ProtectLib` wrappers which will make them accessible for later phases.

This way, we may support arbitrary complex type parameters without implementing complex type parsing for arguments.

wsnyder, 2024-08-05:
Sorry missed replying earlier, but I agree what you suggest with putting types into the generated file seems preferrable.

wsnyder, 2024-08-21:
Thanks for the significant efforts on fixing this.

closed 2024-08-21 by wsnyder