verilator/verilator #5312: Internal Error: Program received signal SIGABRT, Aborted.

luizademelo, 2024-07-29:
When running "verilator --cc --debug --gdbbt" on this file 
[32247_ivtest_ivltests_pr2350934b.txt](https://github.com/user-attachments/files/16413078/32247_ivtest_ivltests_pr2350934b.txt)
Verilator throws this error: 
```
%Error: Internal Error: database/32247_ivtest_ivltests_pr2350934b.v:8:27: ../V3Broken.cpp:168: Broken link in node (or something without maybePointedTo): 'm_taskp && !m_taskp->brokeExists()' @ ./V3Ast__gen_impl.h:3566
                                                                        : ... note: In instance 'top.a'
-node: FUNCREF 0x555556f75b80 <e1225> {d8bb} @dt=0@  f -> .=block FUNC 0x555556f63b00 <e1441#> {d31bc} @dt=0@  f
    8 |     initial dummy = block.f(0);
      |                           ^
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
- V3Ast.cpp:1323:     Dumping obj_dir/V__0332247_ivtest_ivltests_pr2350934b_990_final.tree
- V3StatsReport.cpp:231:statsReport: 
%Error: Internal Error: Aborting since under --debug

Program received signal SIGABRT, Aborted.
```
I'm using this verilator version: 
```
Verilator 5.027 devel rev v5.026-29-ge3e7b3f2b
```
I'm running this on WSL2 Ubuntu. 
This program works fine when running with Icarus Verilog.

wsnyder, 2024-08-05:
Perhaps you'd be willing to look into a pull request to fix this?

wsnyder, 2024-11-25:
Related to #4769

closed 2026-09-29 by wsnyder