verilator/verilator #5314: Unable to iterate over hierarchy using VPI

Shatur, 2024-07-29:
In version 5.022 I was able to iterate over hierarchy by creating the following iterator:
```cpp
vpiHandle module_iter = vpi_iterate(vpiModule, root_handle);
```
But after update to 5.022 (also happens on 5.024) this line returns `nullptr`.
I also noticed that `Verilated::scopesDump()` on 5.022 includes to level modules:
```
  scopesDump:
    SCOPE 0x72870bd3aec0: TOP.aes_core
    SCOPE 0x72870bd3aef8: TOP.aes_core._552_
       VAR 0x5b20f2061d48: ZN
       VAR 0x5b20f2061dc8: cell_kind
...
```
But after 5.024 it looks like this (no `TOP.aes_core`):
```
  scopesDump:
    SCOPE 0x782868622ec0: TOP.aes_core._552_
       VAR 0x5b2088ce9158: ZN
       VAR 0x5b2088ce91d8: cell_kind
...
```

I looked at changes and I suspect that https://github.com/verilator/verilator/pull/4918 or https://github.com/verilator/verilator/pull/4965 could cause it.

Shatur, 2024-07-31:
@AndrewNolte maybe you have an idea about what is going on?

AndrewNolte, 2024-07-31:
What is `root_handle` here? And _552_ ? `test_regress/t/t_vpi_dump.cpp` iterates at the top level, and seems like it does it correctly with `TestVpiHandle it = vpi_iterate(vpiModule, NULL);`. Do you want to make a test case, and maybe bisect to see which one broke it?

Shatur, 2024-08-01:
Ah, `root_handle` is nullptr.

Hm... Looks like `test_regress/t/t_vpi_dump.cpp` uses `/* verilator public_on */`, while I mark specific outputs as true. Like this:
```
module AND2_X1 (
      input wire A1
    , input wire A2
    , output wire ZN /* verilator public_flat_rd */
);
    assign ZN = (A1 & A2);
endmodule
```

I suspect that this causes the issue.

Shatur, 2024-08-01:
@AndrewNolte  [here](https://github.com/user-attachments/files/16454818/verilator_bug.zip) is a minimal example to reproduce the issue. If I run it with Verilator 5.024+, nothing will be printed to the output. But with Verilator 5.022 and below you will see `TOP.top.sub.clk` in the output (because it marked as `/* verilator public_flat_rd */`).

Shatur, 2024-09-06:
@AndrewNolte sorry for the ping, but could you please take a look? Since you the author of the PR that caused the regression.

AndrewNolte, 2024-09-06:
Can you put up the reproducer as a test case draft pull request? And have you found workarounds like making the top level public?

Shatur, 2024-09-18:
@AndrewNolte me and @gephaistos looked into it. 290b313 is the one that causes the issue.

Reverting this change fixes the example I linked in https://github.com/verilator/verilator/issues/5314#issuecomment-2262700774. I.e. `TOP.top.sub.clk` becomes visible.
Also adding something like `localparam my_param /* verilator public_flat_rd */ = 1;` to the `top` module without reverting 290b313 makes both `TOP.top.sub.clk` and `TOP.top.cell_kind` visible. Maybe this information will help you to debug the issue.

Please, take a look at the draft PR with the test case based on the example that reproduces the issue: #5458.
But for some reason reverting 290b313 for this PR causes a segfault. We are not sure why it happens with the test, but works with the example.

PR #5731 (2025-01-13): Fix hierarchy iteration
Current comparison of `prd_pos` with `dot_pos` doesn't take the case of `prd_pos` being `string::npos` into account.

For example: in `TOP__DOT__top__DOT__sub`, `prd_pos == string::npos`. `string::npos` is defined to be -1, but it's unsigned (https://en.cppreference.com/w/cpp/string/basic_string/npos), meaning -1 is converted to the largest integer. This results with the `else` case, which is incorrect.

In addition, we removed the misleading comment.

Fixes #5314.
Ping @AndrewNolte.
closed 2025-01-13 by wsnyder