verilator/verilator #5316: Warnings causes `make` fails

DenizzzGuzell, 2024-07-30:
Hi, 

If there are warnings in project given below command :
`verilator -f flist --cc --top-module top -no-dump-tree`
exits with error as ;
%Error: Exitting due `n` warning(s)

So there is Vtop.h file created. Is created Vtop.h file can be used and even though makefile fails. It represents there are some warnings but no error so .h file can be used. Or Vtop.h is not properly generated due to makefile fail and to verilate a project it have to get zero warnings. 

Failed Makefile really make me think that due to warnings project is not properly verilated and Vtop.h file is not correct. Can you clarify me ?

Thank you

wsnyder, 2024-07-30:
If verilator exits with errors (even if %Error due to warning) the output files should not be used.

closed 2024-07-30 by wsnyder

abdullahyildiz, 2024-08-05:
@wsnyder,
According to [Verilator Documentation](https://verilator.org/guide/latest/warnings.html), some of the warnings can be ignored. Can we safely skip this error if Verilator stops due to the respective warnings?

wsnyder, 2024-08-05:
If it prints "%error" including the line about too many warnings you cannot use outputs.

If you are using -Wno-error, that line will not be printed.  Note some warnings are important as indicate simulation behavior may vary, so I don't recommend blindly ignoring warnings.