verilator/verilator #5319: unique case with --assert falsely failing with multiple matches

VarunKoyyalagunta, 2024-08-01:
I created a test here -
https://github.com/VarunKoyyalagunta/verilator/compare/2f5c58b345eca4bf81cace78e9a1d867fa579eef...bug/unique-case-default

This fails on current master (2f5c58b345eca4bf81cace78e9a1d867fa579eef)
```
[0] %Error: t_unique_case_many.v:215: Assertion failed in top.t.get_hit: unique case, but multiple matches found for '23'h7fffff'
```

verilator appears to do an `if(vlwide_type_var)` to qualify the assertion. But that `if` will always evaluate to `true`, as `VlWide`'s operator converts that to a pointer.

```
VlWide<4>/*127:0*/ __Vtemp_72;
__Vtemp_72[0U] = __Vtemp_38[0U];
 __Vtemp_72[1U] = __Vtemp_38[1U];
__Vtemp_72[2U] = /* snip */
if (__Vtemp_72) {
                                 if (vlSymsp->_vm_contextp__->assertOnGet(32)) {
                                     VL_WRITEF_NX("[%0t] %%Error: t_unique_case_many.v:215: Assertion failed in %Nt.get_hit: unique case, but multiple matches found for '23'h%x'\n",0,
                                                  64,
                                                  VL_TIME_UNITED_Q(1),
                                                  -12,
                                                  vlSymsp->name(),
                                                  23,
                                                  __Vfunc_t__DOT__get_hit__0__a);
                                     VL_STOP_MT("t/t_unique_case_many.v", 215, "");
                                 }
                             }
```

VarunKoyyalagunta, 2024-08-01:
This makes the unit test pass

```
diff --git a/include/verilated_types.h b/include/verilated_types.h
index fa99d32e5..a6107b797 100644
--- a/include/verilated_types.h
+++ b/include/verilated_types.h
@@ -447,6 +447,12 @@ struct VlWide final {
         }
         return false;
     }
+    operator bool() VL_PURE {
+        for (size_t i = 0; i < T_Words; ++i) {
+            if (m_storage[i]) return true;
+        }
+        return false;
+    }

     // METHODS
     const EData& at(size_t index) const { return m_storage[index]; }
 ```

wsnyder, 2024-08-06:
Thanks for your work fixing this.

PR #5324 (2024-08-06): Fix assert on wide expression (#5319)
Fix #5319
closed 2024-08-06 by wsnyder