verilator/verilator #5325: Internal Error instead of normal error for same specparam in different generate-for loop

Rubiczhang, 2024-08-02:
Hello, I get Internal Error when compile such verilog code:
```verilog
genvar i;
generate
//cvt kbd_code_buf, and put result into kbd_segcode
    for(i = 0; i < 2; i++)
    begin: gen_cvt
        hex7seg cvt(
            .b(kbd_code_buf[i*4 + 3: i*4]),
            .en(seg_show_en[0]),
            .out(kbd_segcode_buf[0][i])
        );

    end
endgenerate

generate 
//cvt kbd_code_buf[1], and put result into kbd_segcode
    for(i = 0; i < 2; i++)
    begin: gen_cvt
        hex7seg cvt(
            .b({4'hf, 4'h0}[i*4 + 3: i*4]),
            .en(seg_show_en[1]),
            .out(kbd_segcode_buf[1][i])
        );
    end
endgenerate

```
And after modify the second gen_cvt to gen_cvt_higher, the verilog file can be compiled successfully.
I think this should be a normal error report but not Internal Error.

I'm using Verilator 5.008 2023-03-04 rev v5.008 ,and not sure whether this issue is fixed in the newest version.

wsnyder, 2024-08-05:
Thanks for the report, does seem still present.

If you'd like to look at a pull to fix the error it would be appreciated.

wsnyder, 2025-03-09:
Got fixed somewhere earlier.

closed 2025-03-09 by wsnyder