verilator/verilator #5327: Question: can UNDRIVEN be made more useful

paul-demo, 2024-08-02:
It seems that this doesn't flag as an UNDRIVEN warning even though clk is undriven. Is it possible for Verilator's linter to determine that the clk signal is undriven in `a` and is not a port of module `a` and is not driven in `a` so any uses of it further down in the hierarchy must be undriven?

```verilog
module a (
  output logic x
);
  logic clk;

  b b_inst (.clk, .x)

endmodule

module b (
  input logic clk,
  output logic x
);

endmodule
```

wsnyder, 2024-08-02:
I get the warnings so I don't understand. Please check the example or your verilator version, or otherwise better describe the issue, thanks.

```
%Warning-UNDRIVEN: /svaha/wsnyder/d/a.v:4:8: Signal is not driven: 'clk'
%Warning-UNUSEDSIGNAL: /svaha/wsnyder/d/a.v:8:14: Signal is not used: 'clk'

```

paul-demo, 2024-08-02:
Very odd. Here's a stripped down version of a real design that I actually have simulated now, and it shows the same issue:

```systemverilog
// file: tb.sv
module tb ();

  logic clk;
  logic reset;
  logic [31:0] tick;

  dut_wrap dut_wrap_i (
      .clk  (clk),
      .reset(reset),
      .tick (tick)
  );

endmodule
```

```systemverilog
// file: dut_wrap.sv
module dut_wrap (
    input logic clk,
    input logic reset,
    input logic [31:0] tick
);
  // Note: clk is dangling input; clk2 is undriven.
  logic clk2;

  dut dut_i (
      .clk  (clk2),
      .reset(reset),
      .tick (tick)
  );

endmodule
```

```systemverilog
// file: dut.sv
module dut (
    input logic clk,
    input logic reset,
    input logic [31:0] tick
);

  logic [31:0] x = '0;

  always_ff @(posedge clk) begin
    if (reset) begin
      x <= '0;
    end else begin
      x <= tick;
    end
  end

endmodule
```

The slightly redacted command line call looks like this:
```bash
Running command:
  perl
  /opt/homebrew/bin/verilator
  -cc
  --exe
  -Mdir
  {obfuscated_path}/build
  -DCOCOTB_SIM=1
  --top-module tb
  --vpi
  --public-flat-rw
  --prefix Vtop
  -o tb
  -LDFLAGS '-Wl,-rpath,{python_site_package_path}/cocotb/libs -L{python_site_package_path}/cocotb/libs -lcocotbvpi_verilator'
  --lint-only
  -I{obfuscated_path}
  -CFLAGS -Wno-deprecated
  --trace
  --trace-fst
  --trace-structs
  --timescale 1ns/1ps
  -Wwarn-UNDRIVEN
  {python_site_package_path}/cocotb/share/lib/verilator/verilator.cpp
  {obfuscated_path}/dut_wrap.sv
  {obfuscated_path}/dut.sv
  {obfuscated_path}/tb.sv 

in directory {obfuscated_path}/build
V e r i l a t i o n   R e p o r t: Verilator 5.024 2024-04-05 rev UNKNOWN.REV
Verilator: Built from 0.032 MB sources in 4 modules, into 0.016 MB in 5 C++ files needing 0.000 MB
Verilator: Walltime 0.004 s (elab=0.000, cvt=0.001, bld=0.000); cpu 0.000 s on 1 threads INFO: Running 
```

Could it be something to do with the position of `-Wwarn-UNDRIVEN`? Should it be included in `-CFLAGS'?

paul-demo, 2024-08-02:
Looking through each of those, I guess I've answered my own question: if I remove `--public-flat-rw`, it does show UNDRIVEN warnings. Unfortunately that one is provided automatically by Cocotb, but I'll try to make a workaround to remove that one when linting.

closed 2024-08-02 by paul-demo