verilator/verilator #5330: Trying to create array of strings causes Verilator to segfault, not sure why

notabelincoln, 2024-08-06:
Thanks for taking the time to report this.

**Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)**
The issue is when I code in an array of strings like this:
```
output reg [`STR_ARRAY_MAX(`OUTPUT_STR_LEN):0] instruction_str [(`NUM_STAGES - 1):0];
```
I get the following error
```
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /opt/verilator/share/verilator/bin/verilator_bin --Wall --trace --exe --build -j 20 --Mdir verilator -I../../hw/rtl/inc -I../../hw/rtl/src -Irtl/inc -Irtl/src -Isim/inc --top-module tb_top --cc ../../hw/rtl/src/risc_v_alu_controller.v ../../hw/rtl/src/risc_v_alu.v ../../hw/rtl/src/risc_v_register_file.v ../../hw/rtl/src/risc_v_instruction_memory.v ../../hw/rtl/inc/risc_v_alu.vh ../../hw/rtl/inc/risc_v_instructions.vh rtl/src/tb_function_decoder.v rtl/src/tb_immediate_decoder.v rtl/src/tb_register_decoder.v rtl/src/tb_top.v rtl/inc/tb_strings.vh sim/src/main.cpp
make: *** [Makefile:58: Vtb_top.mk] Error 255
```
But when I split the array into two separate variables like this:
```
output reg [`STR_ARRAY_MAX(`OUTPUT_STR_LEN):0] instruction_str0;
output reg [`STR_ARRAY_MAX(`OUTPUT_STR_LEN):0] instruction_str1;
```
It compiles normally without warning. I write to instruction string variables using ```$sformat```. In theory it takes up the same amount of space. It doesn't matter how large either "dimension" of the array is, it always segfaults. Perhaps it has something to do with writing strings to array elements?

**What 'verilator' command line do we use to run your example?**
```
verilator --Wall  \
        --trace \
        --exe \
        --build \
        -j `nproc` \
        --Mdir verilator \
        -I../../hw/rtl/inc \
        -I../../hw/rtl/src \
        -Irtl/inc \
        -Irtl/src \
        -Isim/inc \
        --top-module tb_top \
        --cc \
         ../../hw/rtl/src/risc_v_alu_controller.v  ../../hw/rtl/src/risc_v_alu.v  ../../hw/rtl/src/risc_v_register_file.v  ../../hw/rtl/src/risc_v_instruction_memory.v \
        ../../hw/rtl/inc/risc_v_alu.vh ../../hw/rtl/inc/risc_v_instructions.vh \
        rtl/src/tb_function_decoder.v rtl/src/tb_immediate_decoder.v rtl/src/tb_register_decoder.v rtl/src/tb_top.v \
        rtl/inc/tb_strings.vh \
        sim/src/main.cpp
```

**What 'verilator --version' are you using?  Did you try it with the git master version?**
```
Verilator 5.027 devel rev v5.026-101-gf4cb2c8cf
```

**What OS and distribution are you using?**
WSL2 Debian Bookworm on Windows 10

**May we assist you in trying to fix this in Verilator yourself?**
Yes

(Please avoid attaching screenshots that show text - you can convert images to text using e.g. https://ocr.space)

wsnyder, 2024-08-06:
Can you please attach a complete example that shows the crash? You can run with --debug and look at the .vpp file verilator creates to also simplify away the preprocessor macros. Thanks.

notabelincoln, 2024-08-07:
> Can you please attach a complete example that shows the crash?

Do you mean a complete output log? If so, I can provide one here:
```
verilator --Wall --debug \
        --trace \
        --exe \
        --build \
        -j `nproc` \
        --Mdir verilator \
        -I../../hw/rtl/inc \
        -I../../hw/rtl/src \
        -Irtl/inc \
        -Irtl/src \
        -Isim/inc \
        --top-module tb_top \
        --cc \
         ../../hw/rtl/src/risc_v_alu_controller.v  ../../hw/rtl/src/risc_v_alu.v  ../../hw/rtl/src/risc_v_register_file.v  ../../hw/rtl/src/risc_v_instruction_memory.v \
        ../../hw/rtl/inc/risc_v_alu.vh ../../hw/rtl/inc/risc_v_instructions.vh \
        rtl/src/tb_function_decoder.v rtl/src/tb_immediate_decoder.v rtl/src/tb_register_decoder.v rtl/src/tb_top.v \
        rtl/inc/tb_strings.vh \
        sim/src/main.cpp
export VERILATOR_ROOT='/opt/verilator/share/verilator'
Starting Verilator 5.027 devel rev v5.026-101-gf4cb2c8cf
- Verilator.cpp:648:  Option --verilate: Start Verilation
- V3File.cpp:234:        --check-times failed: different command line
- V3ParseImp.cpp:292: parseFile: verilated_std
  Preprocessing /opt/verilator/share/verilator/include/verilated_std.sv
- V3PreShell.cpp:143:     Reading /opt/verilator/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:372: Lexing /opt/verilator/share/verilator/include/verilated_std.sv
- V3ParseImp.cpp:292: parseFile: risc_v_alu_controller
  Preprocessing ../../hw/rtl/src/risc_v_alu_controller.v
- V3PreShell.cpp:143:     Reading ../../hw/rtl/src/risc_v_alu_controller.v
- V3PreShell.cpp:143:     Reading ../../hw/rtl/inc/risc_v_instructions.vh
- V3PreShell.cpp:143:     Reading ../../hw/rtl/inc/risc_v_alu.vh
- V3ParseImp.cpp:358: Writing all preprocessed output to verilator/Vtb_top__inputs.vpp
- V3ParseImp.cpp:372: Lexing ../../hw/rtl/src/risc_v_alu_controller.v
- V3ParseImp.cpp:292: parseFile: risc_v_alu
  Preprocessing ../../hw/rtl/src/risc_v_alu.v
- V3PreShell.cpp:143:     Reading ../../hw/rtl/src/risc_v_alu.v
- V3PreShell.cpp:143:     Reading ../../hw/rtl/inc/risc_v_alu.vh
- V3ParseImp.cpp:372: Lexing ../../hw/rtl/src/risc_v_alu.v
- V3ParseImp.cpp:292: parseFile: risc_v_register_file
  Preprocessing ../../hw/rtl/src/risc_v_register_file.v
- V3PreShell.cpp:143:     Reading ../../hw/rtl/src/risc_v_register_file.v
- V3ParseImp.cpp:372: Lexing ../../hw/rtl/src/risc_v_register_file.v
- V3ParseImp.cpp:292: parseFile: risc_v_instruction_memory
  Preprocessing ../../hw/rtl/src/risc_v_instruction_memory.v
- V3PreShell.cpp:143:     Reading ../../hw/rtl/src/risc_v_instruction_memory.v
- V3PreShell.cpp:143:     Reading ../../hw/rtl/inc/risc_v_instructions.vh
- V3ParseImp.cpp:372: Lexing ../../hw/rtl/src/risc_v_instruction_memory.v
- V3ParseImp.cpp:292: parseFile: risc_v_alu
  Preprocessing ../../hw/rtl/inc/risc_v_alu.vh
- V3PreShell.cpp:143:     Reading ../../hw/rtl/inc/risc_v_alu.vh
- V3ParseImp.cpp:372: Lexing ../../hw/rtl/inc/risc_v_alu.vh
- V3ParseImp.cpp:292: parseFile: risc_v_instructions
  Preprocessing ../../hw/rtl/inc/risc_v_instructions.vh
- V3PreShell.cpp:143:     Reading ../../hw/rtl/inc/risc_v_instructions.vh
- V3ParseImp.cpp:372: Lexing ../../hw/rtl/inc/risc_v_instructions.vh
- V3ParseImp.cpp:292: parseFile: tb_function_decoder
  Preprocessing rtl/src/tb_function_decoder.v
- V3PreShell.cpp:143:     Reading rtl/src/tb_function_decoder.v
- V3PreShell.cpp:143:     Reading ../../hw/rtl/inc/risc_v_instructions.vh
- V3PreShell.cpp:143:     Reading rtl/inc/tb_strings.vh
- V3ParseImp.cpp:372: Lexing rtl/src/tb_function_decoder.v
- V3ParseImp.cpp:292: parseFile: tb_immediate_decoder
  Preprocessing rtl/src/tb_immediate_decoder.v
- V3PreShell.cpp:143:     Reading rtl/src/tb_immediate_decoder.v
- V3PreShell.cpp:143:     Reading ../../hw/rtl/inc/risc_v_instructions.vh
- V3ParseImp.cpp:372: Lexing rtl/src/tb_immediate_decoder.v
[... 57 lines cut ...]
        {7'b0000011}: $sformat(instruction_str[0], "%s %s,%s(%s)", function_str[0], rd_str[0], immediate_str[0], rs1_str[0]);
        {7'b0100011}: $sformat(instruction_str[0], "%s %s,%s(%s)", function_str[0], rs2_str[0], immediate_str[0], rs1_str[0]);
        {7'b1100011}: $sformat(instruction_str[0], "%s %s,%s,%s", function_str[0], rs1_str[0], rs2_str[0], immediate_str[0]);
        {7'b0110111}: $sformat(instruction_str[0], "%s %s,%s", function_str[0], rd_str[0], immediate_str[0]);
        {7'b0010111}: $sformat(instruction_str[0], "%s %s,%s", function_str[0], rd_str[0], immediate_str[0]);
        {7'b1101111}: $sformat(instruction_str[0], "%s %s,%s", function_str[0], rd_str[0], immediate_str[0]);
        default: instruction_str[0] = {((36) * 8){1'b0}};
        endcase
end

always @(*) begin
        case ({instruction[1][6:0]})
        {7'b0110011}: $sformat(instruction_str[1], "%s %s,%s,%s", function_str[1], rd_str[1], rs1_str[1], rs2_str[1]);
        {7'b0010011}: $sformat(instruction_str[1], "%s %s,%s,%s", function_str[1], rd_str[1], rs1_str[1], immediate_str[1]);
        {7'b1100111}: $sformat(instruction_str[1], "%s %s,%s+%s", function_str[1], rd_str[1], rs1_str[1], immediate_str[1]);
        {7'b0000011}: $sformat(instruction_str[1], "%s %s,%s(%s)", function_str[1], rd_str[1], immediate_str[1], rs1_str[1]);
        {7'b0100011}: $sformat(instruction_str[1], "%s %s,%s(%s)", function_str[1], rs2_str[1], immediate_str[1], rs1_str[1]);
        {7'b1100011}: $sformat(instruction_str[1], "%s %s,%s,%s", function_str[1], rs1_str[1], rs2_str[1], immediate_str[1]);
        {7'b0110111}: $sformat(instruction_str[1], "%s %s,%s", function_str[1], rd_str[1], immediate_str[1]);
        {7'b0010111}: $sformat(instruction_str[1], "%s %s,%s", function_str[1], rd_str[1], immediate_str[1]);
        {7'b1101111}: $sformat(instruction_str[1], "%s %s,%s", function_str[1], rd_str[1], immediate_str[1]);
        default: instruction_str[1] = {((36) * 8){1'b0}};
        endcase
end
```

wsnyder, 2024-08-07:
Please provide verilog that shows the issue that can be inputted exactly to Verulator to show it.  No "..." you have above as that is not legal verilog.

notabelincoln, 2024-08-07:
Please see the simplified example of the issue-producing code below. For simplicity, I've added a `` `define `` for toggling between the broken and working versions:
```
`timescale 1 ns / 1 ps

//`define USE_BROKEN
`define NUM_STAGES (2)
`include "tb_strings.vh"

module tb_top(
        clk,
        pc,
        instruction,
`ifdef USE_BROKEN
        instruction_str
`else
        instruction_str0,
        instruction_str1
`endif /* USE_BROKEN */
);

input wire clk;

output reg [31:0] pc [(`NUM_STAGES - 1):0];
output reg [31:0] instruction [(`NUM_STAGES - 1):0];
`ifdef USE_BROKEN
output reg [`STR_ARRAY_MAX(`OUTPUT_STR_LEN):0] instruction_str [(`NUM_STAGES - 1):0];
`else
output reg [`STR_ARRAY_MAX(`OUTPUT_STR_LEN):0] instruction_str0;
output reg [`STR_ARRAY_MAX(`OUTPUT_STR_LEN):0] instruction_str1;
`endif /* USE_BROKEN */

integer i;

always @(posedge clk) begin
        pc[0] <= pc[0] + 32'd4;
        instruction[0] <= instruction[0] + 32'd4;
        for (i = 1; i < `NUM_STAGES; i = i + 1) begin
                pc[i] <= pc[i - 1];
                instruction[i] <= instruction[i - 1];
        end
end

// Put together a final string
`ifdef USE_BROKEN
always @(*) begin
        case (instruction[0][0])
        1'b0: $sformat(instruction_str[0], "%s", "Hello");
        1'b1: $sformat(instruction_str[0], "%s", "Goodbye");
        endcase
end

always @(*) begin
        case (instruction[1][0])
        1'b0: $sformat(instruction_str[1], "%s", "Hello");
        1'b1: $sformat(instruction_str[1], "%s", "Goodbye");
        endcase
end
`else
always @(*) begin
        case (instruction[0][0])
        1'b0: $sformat(instruction_str0, "%s", "Hello");
        1'b1: $sformat(instruction_str0, "%s", "Goodbye");
        endcase
end

always @(*) begin
        case (instruction[1][0])
        1'b0: $sformat(instruction_str1, "%s", "Hello");
        1'b1: $sformat(instruction_str1, "%s", "Goodbye");
        endcase
end
`endif /* USE_BROKEN */
endmodule
```
I am also showing the corresponding .vpp file:
```
`begin_keywords "1800-2023"
`line 1 "rtl/src/tb_top.v" 1
`timescale 1 ns / 1 ps

`line 3 "rtl/src/tb_top.v" 0

`line 5 "rtl/src/tb_top.v" 0
`line 1 "rtl/inc/tb_strings.vh" 1

`line 2 "rtl/inc/tb_strings.vh" 0

`line 4 "rtl/inc/tb_strings.vh" 0

`line 9 "rtl/inc/tb_strings.vh" 0

`line 14 "rtl/inc/tb_strings.vh" 0
`line 5 "rtl/src/tb_top.v" 2

`line 7 "rtl/src/tb_top.v" 0
module tb_top(
        clk,
        pc,
        instruction,

`line 14 "rtl/src/tb_top.v" 0
        instruction_str0,
        instruction_str1

);

`line 19 "rtl/src/tb_top.v" 0
input wire clk;

`line 21 "rtl/src/tb_top.v" 0
output reg [31:0] pc [((2) - 1):0];
output reg [31:0] instruction [((2) - 1):0];

`line 26 "rtl/src/tb_top.v" 0
output reg [(((36) * 8) - 1):0] instruction_str0;
output reg [(((36) * 8) - 1):0] instruction_str1;

`line 30 "rtl/src/tb_top.v" 0
integer i;

`line 32 "rtl/src/tb_top.v" 0
always @(posedge clk) begin
        pc[0] <= pc[0] + 32'd4;
        instruction[0] <= instruction[0] + 32'd4;
        for (i = 1; i < (2); i = i + 1) begin
                pc[i] <= pc[i - 1];
                instruction[i] <= instruction[i - 1];
        end
end

`line 42 "rtl/src/tb_top.v" 0

`line 51 "rtl/src/tb_top.v" 0

`line 58 "rtl/src/tb_top.v" 0
always @(*) begin
        case (instruction[0][0])
        1'b0: $sformat(instruction_str0, "%s", "Hello");
        1'b1: $sformat(instruction_str0, "%s", "Goodbye");
        endcase
end

`line 65 "rtl/src/tb_top.v" 0
always @(*) begin
        case (instruction[1][0])
        1'b0: $sformat(instruction_str1, "%s", "Hello");
        1'b1: $sformat(instruction_str1, "%s", "Goodbye");
        endcase
end

endmodule

`line 75 "rtl/src/tb_top.v" 0
```

closed 2024-08-08 by wsnyder