verilator/verilator #5336: Coverage: emit lossless lcov file

AndrewNolte, 2024-08-06:
Branch info is supported by lcov, see [here](https://xujihui1985.github.io/test/2015/10/06/lcovinfo) for explanations 

We can modify the coverage annotation to fix confusing lcov data like https://github.com/verilator/verilator/issues/4691. The branch coverage view isn't great on the default lcov genhtml, but it can be interpreted by other viewers like [gcovr](https://stackoverflow.com/questions/62936628/how-to-improve-the-branch-coverage-using-gcovr-tool), or a [vscode extension](https://marketplace.visualstudio.com/items?itemName=ryanluker.vscode-coverage-gutters)

wsnyder, 2024-08-06:
Agreed. Hoping this is a suggestion likely to lead to a pull request ;)

AndrewNolte, 2024-08-19:
I'm trying to make the changes, but it was really annoying that they weren't being propogated to the final binary. I put a clear compile error in `src/VlcSource.h` to test it, and when I built, it produced a new `verilator_coverage_dbg_bin` successfully. Something must be wrong with the build dependencies?

wsnyder, 2024-08-20:
> I'm trying to make the changes, but it was really annoying that they weren't being propogated to the final binary. I put a clear compile error in src/VlcSource.h to test it, and when I built, it [...]

I did a build, then for each vlc .cpp and header checked that a touch of that file did a rebuild. It all worked. So not sure what's up with your side. Note the -MMD flag should be auto-building the dependencies. One thing to be careful is if using NFS that your server and client clocks are synchronized.

AndrewNolte, 2024-08-20:
I was doing `make -j`, is that not the right command to run?

wsnyder, 2024-08-20:
make -j should be fine. That's what I was using too.

closed 2024-09-11 by wsnyder