verilator/verilator #5341: xor reduction: packed vector used instead of value in comparison

tilir, 2024-08-07:
Reproduction: https://github.com/tilir/sverilog/blob/master/combinational/repro_bug.sv

```
$ verilator -cc --timing --build --main --exe repro_bug.sv
....
error: no match for ‘operator!=’ (operand types are ‘VlUnpacked<unsigned char, 4>’ and ‘IData’ {aka ‘unsigned int’})
if (VL_UNLIKELY((vlSelf->bench__DOT__a != (IData)(vlSelf->bench__DOT__yeta)))) {
```

```
$ verilator --version
Verilator 5.020 2024-01-01 rev (Debian 5.020-1)
```

Looks like verilator instead of generating comparison y vs yeta, generates a vs yeta.

> May we assist you in trying to fix this in Verilator yourself?

Probably I may try to fix this bug on my own, but I need some guidance on what is the best way to fix.

wsnyder, 2024-08-08:
This is not legal code, you cannot slice a packed array that way.  Verilator needs to add a verilation-time error to flag this.

udif, 2024-08-09:
The problem is that all his code uses unpacked arrays, but he treats them like packed arrays.
Perhaps what could help is catching `a[m:n]` RHS and LHS references when `a` is a packed array.