verilator/verilator #5342: Memory and runtime problem about V3Variable

godblesszhouzhou, 2024-08-08:
I'm simulating a big project related with RISC-V and one problem was found. The V3VariableOrder takes tow much memory and compile time.  After the V3Common::commonAll() is finished, verilator takes about 26G memory, while when V3VariableOrder::orderAll() is run, the memory increases to 320G.  The whole compile time is 3258s, while the compile time for V3Common::commonAll() is 2442s (about 75%).  So I think there must be some problems here.

godblesszhouzhou, 2024-08-08:
By further debugging, the main problem is codes:
`
    for (AstNodeModule* modp = v3Global.rootp()->modulesp(); modp;
         modp = VN_AS(modp->nextp(), NodeModule)) {
        VariableOrder::processModule(modp);
    }
`
So the main problem is probably the function **orderModuleVars**.

godblesszhouzhou, 2024-08-08:
Thanks for suggestions. I have done the debug further. The main problem is function **tspSortVars**.  For my case, the vector varps' size is  1132665 (not very large I think). Before "tspSortVars(varps)" is run, everything is ok and the memory taken is about 26G. While when tspSortVars is run, it take a very long time and the memory nearly increases to 300GB.

gezalore, 2024-08-08:
TSP sort is O(n^2) in space, so this can happen, see #4960.

godblesszhouzhou, 2024-08-08:
Is that a reasonable increase in memory(30GB -> 300GB )？ As for my simulation this part takes about 75% compile time and 90% compile memory, if there are not too many benefits,  it's better to close it.

wsnyder, 2024-08-08:
I suggest we  add a -mno-variable-order-tsp  flag to fall back to the nonmtask algorithm. @godblesszhouzhou  perhaps you'd be willing to make a pull to do that?

wsnyder, 2024-08-08:
Or maybe with that flag just group by mtask number then historical.  False sharing will kill performance otherwise.

godblesszhouzhou, 2024-08-09:
I have read this part of codes and made some some tests further. The main time and memory cost is in function **V3TSP::tspSort(states, &sortedStates)**. The **vector varps' size is 1132665** and **vector states' size is 58324**. As the V3TSP::tspSort's  time cost and memory cost are both O(N^2), it is reasonable to get this result. @gezalore . Then I will try to fall back to nonmtask algorithm to make a test whether it is ok for my case. @wsnyder

gezalore, 2025-09-10:
Referring back to #4960 again, I think TSP sort is only used to minimize dcache false sharing. It would be worth trying to just insert padding between groups of variables with different mtask affinity sets so they don't share cache lines, then we could delete the expensive TSP sort entirely.

wsnyder, 2025-09-10:
I believe there are three things to help both capacity and misses:
1. Make sure different mtask'ed variables are padded to different lines. (easy)
2. Make sure multiple variables pack into a cache line (part of V3Emit). (easy)
3. Try to get likely near-tenporally accessed data together. (NP complete)

I would guestimate this alg being good vs trivial would swing performance 10%ish.

Like most NP problems we'll want to improve the alg to cheat a bit from optimal for compile performance.

gezalore, 2025-09-10:
Agreed, and If I read it right, the TSP sort achieves none of these.