verilator/verilator #5345: Indexed vector part-select behaves incorrectly on arrays that have nonzero starting index range

jamesrbailey, 2024-08-08:
Using the indexed vector part-select mechanism to slice an array that has been defined with an non-zero-starting index range behaves incorrectly. I am suspicious that verilator may make some assumptions that arrays start from zero index under certain circumstances.

I am running Verilator 5.027 devel rev v5.026-107-ge6fe367bd

I have created two small testcases to show this behaviour.

The first: I can elicit a unexpected compile failure giving a bit more insight into the failure by starting the vectored slice at a negative index:

    module test;
        int A[-10:10];
        int slice[3];
    
        initial begin
            foreach(A[i])
                A[i] = i;
    
            $display("Contents of A");
            foreach(A[i])
                $display(i, A[i]);
    
            $display("Contents of slice");
            slice = A[-1 +: 3];
            foreach(slice[i])
                $display(i, slice[i]);
    
            $finish();
        end
    endmodule

Produces this compile failure

    %Error: test2.sv:15:18: Slice selection index '[-11:-9]' outside data type's '[-10:10]'
                          : ... note: In instance 'test'
       15 |         slice = A[-1+:3];
          |

Note that while I should be selecting indices [-1:1] verilator thinks I am selecting [-11:-9]

The second: The wrong part of the array is selected and a functional bug is observed during runtime.

    module test;
    
        int A[-10:10];
        int slice[3];
    
        initial begin
            foreach(A[i])
                A[i] = i;
    
            $display("Contents of A");
            foreach(A[i])
                $display(i, A[i]);
    
            $display("Contents of slice");
            slice = A[0+:3];
            foreach(slice[i])
                $display(i, slice[i]);
    
            $finish();
        end
    
    endmodule

Gives the following result:

    Contents of A
            -10        -10
             -9         -9
             -8         -8
             -7         -7
             -6         -6
             -5         -5
             -4         -4
             -3         -3
             -2         -2
             -1         -1
              0          0
              1          1
              2          2
              3          3
              4          4
              5          5
              6          6
              7          7
              8          8
              9          9
             10         10
    Contents of slice
              0        -10
              1         -9
              2         -8

Note that the code should be selecting indices [0:2] but instead verilator selects [-10:-8]

wsnyder, 2024-08-08:
Indeed. Should be easy to fix, might you make an attempt at a pull?

I'd also suggest your test case have a 

     int B[10:-10];

and a ` -:3` test of A and B.

jamesrbailey, 2024-08-14:
I'm going to _attempt_ a pull request. 

I think I should be probably looking at `WidthSelVisitor::replaceSelPlusMinus`?

From some debug, I gather verilator uses an internal array start index of 0 regardless of what is specified in the Verilog array declaration, and then an offset is applied to the Verilog slice indices. This offset seems to be getting applied to conventional slices ie. [A:B] but not plus/minus slices ie. [A+:WIDTH].

wsnyder, 2024-08-14:
Most likely that function, yes.  It generally does lsb subtracts ok in other cases FWIW.

jamesrbailey, 2024-08-14:
The subtract is missing in this case, and the following seems to fix the issue for me:
```diff
diff --git a/src/V3WidthSel.cpp b/src/V3WidthSel.cpp index ddf3ac719..95abd411d 100644
--- a/src/V3WidthSel.cpp
+++ b/src/V3WidthSel.cpp
@@ -551,8 +551,8 @@ class WidthSelVisitor final : public VNVisitor {
                 // down array: msb/hi -: width
                 // up array:   msb/lo +: width
                 // up array:   lsb/hi -: width
-                const int32_t msb = VN_IS(nodep, SelPlus) ? rhs + width - 1 : rhs;
-                const int32_t lsb = VN_IS(nodep, SelPlus) ? rhs : rhs - width + 1;
+                const int32_t msb = (VN_IS(nodep, SelPlus) ? rhs + width - 1 : rhs) - fromRange.lo();
+                const int32_t lsb = (VN_IS(nodep, SelPlus) ? rhs : rhs 
+ - width + 1) - fromRange.lo();
                 AstSliceSel* const newp = new AstSliceSel{
                     nodep->fileline(), fromp, VNumRange{msb, lsb, fromRange.ascending()}};
                 nodep->replaceWith(newp);

```

I am going to review some of the other SelPlusMinus cases in case they have similar issues, then add testing, and make a pull request.

PR #5387 (2024-08-20): Fix +: and -: unpacked array slicing when array has nonzero low index. (#5345)
Resolves #5345

Subtracts low index of unpacked array from msb & lsb calculations during plus/minus slice replacement. This is necessary when the low index is not zero, and I think was simply missed previously as this may not be a common coding style.

I have updated the existing t_unpacked_slice test so that it defines the source array parameters to have a negative low indices. I believe this should expand test coverage to this particular issue without sacrificing existing coverage. I'm open to this deserving its o [... cut]
closed 2024-08-20 by wsnyder