verilator/verilator #5351: `lint_off -match` includes Errors

sifferman, 2024-08-09:
```
Ubuntu 22.04.4 LTS
Verilator 5.027 devel rev v5.026-4-g833df6d9e
```

Hello. I have been recently experimenting with `lint_off -match`, but I noticed that it suppresses errors, even though the documentation makes it seem like only warnings will be suppressed.

> If -match is set, the linter warnings are matched against this (wildcard) string and are waived in case they match, provided with the rule and file also match.

This often leads to internal Verilator faults.

Should `lint_off -match` never match `EC_FATAL`, `EC_FATALEXIT`, `EC_FATALSRC`, and `EC_ERROR`? I can do a PR for this if you think this is a good idea.

## Example

```bash
# command
verilator --lint-only bad.sv lint.vlt
```

```systemverilog
// bad.sv
module bad;
    initial $readmemh("file.memh", ENUMITEM);
endmodule
```
```systemverilog
// lint.vlt
`verilator_config
lint_off -file "bad.sv" -match "*"
```

```
# Expected output:
%Error: bad.sv:3:36: Can't find definition of variable: 'ENUMITEM'
    3 |     initial $readmemh("file.memh", ENUMITEM);
      |                                    ^~~~~~~~
%Error: Exiting due to 1 error(s)
```

```
# Actual result:
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec verilator_bin --lint-only bad.sv lint.vlt
```

wsnyder, 2024-08-09:
That's definitely wrong. A pull would be very appreciated.

wallento, 2024-08-09:
Oh, yeah, this is entirely right, I didn't test for this unfortunately

PR #5363 (2024-08-11): Prevent lint_off on Errors (#5351)
Fixes #5351.

Happy to make adjustments!