verilator/verilator #5352: Cannot change the lint language standard?

jimmysitu, 2024-08-09:
I am trying to compile a simple model with verilator, 5.0.26

```verilog
module t (
    output q,
    input a,
    input b,
    input s
);

reg q;

always @(*)
    if(s) begin
        q = a;
    end else begin
        q = b;
    end

endmodule
```
And I try `verilator --default-language 1364-2005 --lint-only t.v`
```text
%Error: t.v:8:5: Duplicate declaration of signal: 'q'
               : ... note: ANSI ports must have type declared with the I/O (IEEE 1800-2023 23.2.2.2)
    8 | reg q;
      |     ^
        t.v:2:12: ... Location of original declaration
    2 |     output q,
      |            ^
```
It seems the language std is still checking with 1800-2023.
This module is OK with iverilog, and I think it is compatible with IEEE 1364-2005, 
Anything I missed?

I also try these cmds, but not work
```bash
verilator --language 1364-2005 --lint-only t.v
verilator --language 1364-2005 --cc t.v
```

wsnyder, 2024-08-09:
Those flags for the most part change only key words, Verilator had almost no checks for missing new features on old versions. I would be reluctant to support checking for all of those cases.

However the ANSI port syntax when in 1995 is one of those things that seems worth a lint warning, if you want to make a pull to add that.

closed 2024-08-09 by wsnyder

jimmysitu, 2024-08-09:
Thanks for the explain @wsnyder 
Is there any work round to bypass this error to warning without modified this old style verilog?

wsnyder, 2024-08-09:
Use 1800 standard.