verilator/verilator #5353: Error: Rollover for V3Const::constifyCpp

godblesszhouzhou, 2024-08-09:
I'm simulating a riscv-v project with the version v5.026. When run V3Const::constifyCpp (After V3Exapand::expandAll),  there is an assert: **"Internal Error: ../V3Ast.h:1985::Rollover"**.  Obviously, the problem is that there is a bug in V3Const causing **cloneTree** of some nodes is called repeatedly untill rollover.  I have do some debug. As for  V3Const::constifyCpp, m_docpp = true and m_doNoconst = true, I test by setting "m_docpp = true , m_doNoconst = false" and "m_docpp = false, m_doNoconst = true".  For the two conditions, the result is ok, while when setting "m_docpp = true , m_doNoconst = true", there will be the assert. As there're a lot of codes in V3Const, I haven't thought of a good way to debug.

wsnyder, 2024-08-09:
Did you try it on git master?

What does the '--debug --gdbbt' report?

To hit this there needs to by 4 billion clearTree calls which, yes, suggests something else broken,unless your design is larger then ever seen before.

godblesszhouzhou, 2024-08-13:
For the master, by removing the function **matchiBitOpTree** for AstOr*, the result is ok now. Namely

    TREEOP ("AstOr  {operandShiftSame(nodep)}",         "replaceShiftSame(nodep)");
    TREEOP ("AstXor {operandShiftSame(nodep)}",         "replaceShiftSame(nodep)");
    **TREEOPC("AstAnd {matchBitOpTree(nodep)}", "DONE");** // For this in tag v5.020, memory increased about 3GB.
    **TREEOPC("AstOr  {matchBitOpTree(nodep)}", "DONE");** // remove this line
    TREEOPC("AstXor {matchBitOpTree(nodep)}", "DONE");

Besides, when I test the v5.020, for **REEOPC("AstAnd {matchBitOpTree(nodep)}", "DONE");** , there is also about 3GB memory increase. So, I think there should be a bug in the function matchBitOpTree.

wsnyder, 2024-08-13:
@yTakatsukasa any ideas?

gezalore, 2024-08-13:
Might or might not be helpful, but this patch (which is on master since March), have saved me a lot of memory (~50G -> ~5G). Bit op tree does create some redundant trees, which can be huge and numerous, but are then deleted. Maybe we could avoid creating those trees that get deleted anyway? Unfortunately I can't provide an easy case.
c92ced4ac79a84f1ed8ad9f1050571f6ee0e387b

gezalore, 2024-08-13:
Assuming those involve cloning...

yTakatsukasa, 2024-08-14:
I'm not so sure, one hypothesis is miscounting of ops.
The function calculates the number of ops (AND, OR, XOR...), and adopts the result if the number of ops decreases by the pass.
I'm afraid miscounting may lead to infinite loop of optimization.

If the design or a small test case can be shared, that would really help.

wsnyder, 2024-08-24:
@godblesszhouzhou can you provide a test case please?

godblesszhouzhou, 2024-08-26:
> @godblesszhouzhou can you provide a test case please?

I tried to make a case last week, but it is not a easy thing. As the design is large, it is not easy to find which part of the Ast Tree course this issue. I will try to find a method to find the issue.

wsnyder, 2024-10-09:
@godblesszhouzhou did you find a test case for this?  If not we'll close this until get one. Thanks.

godblesszhouzhou, 2024-10-10:
No. A good debug method for a big design is needed but I have not found one. I think we can close this.

gezalore, 2024-10-10:
Please consider using the new and excellent sv-bugpoint tool:
https://antmicro.com/blog/2024/09/sv-bugpoint-for-improved-debugging-in-sv-tools/

closed 2024-10-10 by wsnyder