verilator/verilator #5358: %Error-UNSUPPORTED -  Unsupported: Default value on module input:

drewranck, 2024-08-09:
Thanks for taking the time to report this.

What would you like added/supported?
--Support for inputs with default values (and no warning on the open connection on the instance)

What 'verilator --version' are you using?  Did you try it with the git master version?
--I'm on 5.027

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)
--I'll follow on with a test for this.

May we assist you in trying to fix this in Verilator yourself?
--With some pointers or hints, I can get started on it.

wsnyder, 2024-08-10:
A pull would be appreciated. This should be fairly straight forward to implement.

First make tests to check the behavior - be sure to also have one that uses an elaboration-time function to choose the default value.  Also test not using the initial value, and overriding it in the instantiating module. Have 2 versions of the test one for inlining on, and another inlining off (look at existing tests). Also have a negative test that tries to default an output - we'll want an error.  

Then modify sources to support this, the parser should put it into the existing initial value of AstVar.  V3Width will need to determine data types of the variable.  Then V3Inline and V3Inst will need to use the default value.

drewranck, 2024-08-12:
Thanks, working on it.

drewranck, 2024-08-13:
Is a default value on an output an error? This may fall into some corner case of supporting an FPGA style flip flop with initial (const) value that so happens to have a shared signal of an output port.

Asking b/c this doesn't appear to be currently checked in Verilator.
V3LinkParse.cpp - "Unsupported: Default value on module input" - specifically checks `nodep->isNonOutput()`.

Maybe I'm not implementing this change correctly, b/c I didn't touch V3Inline, V3Inst, or V3Width. Aside from the section in V3LinkParse.cpp, there's a section in V3LinkCells.cpp where it looks for PINMISSING. I'd obviously not want to warn on a missing input pin if it has a default value, so I added some code there to not warn if the AstPort's matching AstVar (in the module statements) is an Input and has a default value.

wsnyder, 2024-08-13:
I'd think a default value on an output is ok as long as there are no other drivers (there may not be a check for this now), because the unsupported error sort of covered it.

It's certainly possible the V3Inline etc doesn't need changes.  Whatever works ;)

drewranck, 2024-08-14:
Yeah, I'll look into the multi-driver error in this case, or at least try some more examples to see how loose the checking is.

I'll also take a closer look at V3Inline. I thought I could get away with all checks pre-linking, but turns out that didn't work and I did need some tiny changes in V3Width post-linking -- for example pre-linking it was impossible to distinguish between a "module input w/ default value" vs a "module input that later has a concurrent assign statement to it".

wsnyder, 2024-08-15:
Appreciate your work on this.

closed 2024-08-15 by wsnyder