verilator/verilator #5359: Internal Error: txn_class.sv:67:24: ../V3Width.cpp:6979: Node has no type

solomatnikov, 2024-08-10:
```
%Error: txn_class.sv:67:24: new() not expected in this context
                                                     : ... note: In instance 'top'
   67 |       txn_type_t txn = new;
      |                        ^~~
%Error: Internal Error: txn_class.sv:67:24: ../V3Width.cpp:6979: Node has no type
                                                                     : ... note: In instance 'top'
   67 |       txn_type_t txn = new;
      |                        ^~~
```

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
class txn_class; // extends uvm_sequence_item;
   rand txn_type_t   req_txn_type;

...

   static function txn_type_t generate_txn(txn_type_t txn_type);
      txn_type_t txn = new;
//      bit success = txn.randomize() with {
//         req_txn_type == txn_type;
//      };
      bit success = txn.randomize();
      assert(success);
      return txn;
   endfunction // generate_txn

endclass
```

What 'verilator' command line do we use to run your example?

```
verilator  --cc --exe -sv --assert --trace --trace-params --trace-structs --report-unoptflat --no-timing --timescale 1ps/1ps --x-assign unique --unroll-count 256 
```

What 'verilator --version' are you using?  Did you try it with the git master version?

**v5.026-ga0ea96b35**

What OS and distribution are you using?

**Linux 5.14.0-427.24.1.el9_4.x86_64**

solomatnikov, 2024-08-10:
Note: with this there is no internal error:

```
   static function tl2_txn_class generate_txn(txn_type_t txn_type);
      tl2_txn_class txn = new;
```

wsnyder, 2024-08-11:
I will make the error wording better, but we need a complete runable example to see if the error is real or incorrect.

solomatnikov, 2024-08-11:
24b097b228dc4c6638cd28588aacde484000978a fixed internal error on `new` but internal error still happens on incorrect `randomize`:

```
%Error: t/t_class_new_ref_bad.v:16:24: new() assignment not legal to non-class data type 'int'
                                     : ... note: In instance 't'
   16 |       txn_type_t txn = new;
      |                        ^~~
%Error-UNSUPPORTED: t/t_class_new_ref_bad.v:18:25: Unsupported: Member call on object 'VARREF 'txn'' which is a 'REFDTYPE 'txn_type_t''
                                                 : ... note: In instance 't'
   18 |       int success = txn.randomize();
      |                         ^~~~~~~~~
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=5.027
%Error: Internal Error: t/t_class_new_ref_bad.v:18:24: ../V3Width.cpp:6984: Node has no type
                                                     : ... note: In instance 't'
   18 |       int success = txn.randomize();
      |                        ^
(Exec expected to fail, and did.)
%Warning: vlt/t_class_new_ref_bad: Line 5 miscompares; obj_vlt/t_class_new_ref_bad/vlt_compile.log != t/t_class_new_ref_bad.out
F1: %Error-UNSUPPORTED: t/t_class_new_ref_bad.v:18:25: Unsupported: Member call on object 'VARREF 'txn'' which is a 'REFDTYPE 'txn_type_t''
F2: %Error: t/t_class_new_ref_bad.v:17:25: new() cannot copy from non-class data type 'int'
To update reference: HARNESS_UPDATE_GOLDEN=1 {command} or --golden
%Warning: vlt/t_class_new_ref_bad: Missing ok
```

Here is the test case:

```
// DESCRIPTION: Verilator: Verilog Test module
//
// This file ONLY is placed under the Creative Commons Public Domain, for
// any use, without warranty, 2024 by Wilson Snyder.
// SPDX-License-Identifier: CC0-1.0

class Base;
endclass

class Cls extends Base;
   typedef int txn_type_t;  // Bad type

   rand txn_type_t req_txn_type;

   static function txn_type_t generate_txn();
      txn_type_t txn = new;
//      txn_type_t copy = new txn;
      int success = txn.randomize();
      assert(success != 0);
      return txn;
   endfunction

endclass

module t(/*AUTOARG*/);

   initial begin
      Base b = Cls::generate_txn();
      $display("%p", b);

      $write("*-* All Finished *-*\n");
      $finish;
   end

endmodule
```

closed 2024-08-13 by wsnyder