verilator/verilator #5360: `LexerInput` and `LexerOutput` out-of-line definition when building on Mac OS

ChrisKjellqvist, 2024-08-10:
I reproduced this issue on the current commit (as of August 9, 2024) and v5.026 on Mac OSX (Sonoma 14.5).
When building both the `autoconf` and CMake build paths, the compilation of "V3ParseLex.cpp.o" results in the following error.

```
[  4%] Building CXX object src/CMakeFiles/verilator.dir/V3ParseLex.cpp.o
In file included from <path>/verilator/src/V3ParseLex.cpp:30:
<path>/verilator/build/src/V3Lexer_pregen.yy.cpp:8722:18: error: out-of-line definition of 'LexerInput' does not match any declaration in 'V3LexerBase'
int yyFlexLexer::LexerInput( char* buf, int /* max_size */ )
                 ^~~~~~~~~~
<path>/verilator/build/src/V3Lexer_pregen.yy.cpp:8751:19: error: out-of-line definition of 'LexerOutput' does not match any declaration in 'V3LexerBase'
void yyFlexLexer::LexerOutput( const char* buf, int size )
                  ^~~~~~~~~~~
2 errors generated.
```

I resolved this compilation error by modifying the generated code in "V3Lexer_pregen.yy.cpp" to use `size_t` instead of `int` on the lines indicated by the error messages:

```
size_t yyFlexLexer::LexerInput( char* buf, size_t /* max_size */ )
...
void yyFlexLexer::LexerOutput( const char* buf, size_t size )

```

I have not verified the functionality of `verilator` using these fixes, but I don't think it should affect correctness. I am not a user of `bison` or `flex` myself, so I don't know the process by which these type mismatches manifest, but, regardless, this should be fixed.

wsnyder, 2024-08-10:
Thanks for debugging. src/flexfix has a bunch of similar hacks to fix broken flex issues, can you please try editing that, then submit a pull with your fix?

wsnyder, 2024-09-11:
See #5361 - with new flex not seen.

closed 2024-09-11 by wsnyder