verilator/verilator #5362: Verilator with --binary on MacOs

HakamAtassi, 2024-08-11:
Hello!

Hope you're doing well. I've had to resort to working on my MacBook recently. On my primary system, I've been taking full advantage of the new `--binary` feature (great feature btw) which works great on linux. On my M2 MacBook, I'm not sure exactly what is going on, but it seems that at some point during verilation, the tool will  run a file that has an ```std::coroutine_handle``` that clang doesnt seem to provide out of the box, at least not without a flag of some sort. The complete error is: 

```
 ~/Repos/test/ c++ -Os -I.  -MMD -I/opt/homebrew/Cellar/verilator/5.026/share/verilator/include -I/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -DVL_TIME_CONTEXT  -std=gnu++17 -fcoroutines-ts -c -o verilated_timing.o /opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.cpp
In file included from /opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.cpp:23:
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:104:10: error: no template named 'coroutine_handle' in namespace 'std'
    std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
    ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:118:28: error: no template named 'coroutine_handle' in namespace 'std'
    VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
                      ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:208:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
```

I went ahead and performed a few sanity checks to make sure I wasn't using a broken install of verilator or something. Running a simple testbench the old school way has no problems, both for a simple counter design and my actual much larger design. I also built a simple coroutine hello world program using std::couroutine_handle and had no problems running with g++ -std=c++20.

I also tried forcing verilator to use g++ (actual g++, not a g++ call that just links to clang) when verilating, but that was unfortunately not very helpful. I'm not exactly sure what else to try. If anyone has suggestions, please do let me know. Thanks in advance.

kbieganski, 2024-08-12:
Does passing `-CFLAGS -std=c++20` to Verilator help as a workaround?

HakamAtassi, 2024-08-13:
Hey @kbieganski. Sorry for the late response. I was pretty sure I tried that before but I wanted to confirm before I replied. 

I went ahead and tried building with 

```verilator -Wall --cc top_v2.sv --binary --trace --top-module top_v2 -I"$SV_DIR" $WARNINGS -CFLAGS -std=c++20```

which unfortunately results in 
```c++ -Os -I/opt/flex/flex-2.6.4/include -I/opt/bison/bison-3.7.91/share/include  -I.  -MMD -I/opt/homebrew/Cellar/verilator/5.026/share/verilator/include -I/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++20 -DVL_TIME_CONTEXT  -std=gnu++17 -fcoroutines-ts -c -o verilated.o /opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated.cpp
c++ -Os -I/opt/flex/flex-2.6.4/include -I/opt/bison/bison-3.7.91/share/include  -I.  -MMD -I/opt/homebrew/Cellar/verilator/5.026/share/verilator/include -I/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++20 -DVL_TIME_CONTEXT  -std=gnu++17 -fcoroutines-ts -c -o verilated_vcd_c.o /opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_vcd_c.cpp
c++ -Os -I/opt/flex/flex-2.6.4/include -I/opt/bison/bison-3.7.91/share/include  -I.  -MMD -I/opt/homebrew/Cellar/verilator/5.026/share/verilator/include -I/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-c++11-narrowing -Wno-constant-logical-operand -Wno-non-pod-varargs -Wno-parentheses-equality -Wno-shadow -Wno-sign-compare -Wno-tautological-bitwise-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -std=c++20 -DVL_TIME_CONTEXT  -std=gnu++17 -fcoroutines-ts -c -o verilated_timing.o /opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.cpp
In file included from /opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.cpp:23:
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:104:10: error: no template named 'coroutine_handle' in namespace 'std'
    std::coroutine_handle<> m_coro;  // The wrapped coroutine handle
    ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:118:28: error: no template named 'coroutine_handle' in namespace 'std'
    VlCoroutineHandle(std::coroutine_handle<> coro, VlProcessRef process, VlFileLineDebug fileline)
                      ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:208:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:276:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:324:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:374:29: error: no template named 'coroutine_handle' in namespace 'std'
    void await_suspend(std::coroutine_handle<> coro) const { coro.destroy(); }
                       ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:409:37: error: no template named 'coroutine_handle' in namespace 'std'
            void await_suspend(std::coroutine_handle<> coro) {
                               ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:426:14: error: no template named 'coroutine_handle' in namespace 'std'
        std::coroutine_handle<> m_continuation;  // Coroutine to resume after this one finishes
        ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:434:14: error: no type named 'suspend_never' in namespace 'std'
        std::suspend_never initial_suspend() const { return {}; }
        ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:438:14: error: no type named 'suspend_never' in namespace 'std'
        std::suspend_never final_suspend() noexcept;
        ~~~~~^
[... cut ...] mebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.h:473:29: error: no template named 'coroutine_handle' in namespace 'std'
    void await_suspend(std::coroutine_handle<> coro) { m_promisep->m_continuation = coro; }
                       ~~~~~^
/opt/homebrew/Cellar/verilator/5.026/share/verilator/include/verilated_timing.cpp:237:6: error: no type named 'suspend_never' in namespace 'std'```

It seems that the compiler is still having a hard time coroutines during compilation.

kbieganski, 2024-08-13:
Ah, it gets overridden by `-std=gnu++17 -fcoroutines-ts` later. That's unfortunate.

I think this is due to a specific combination of Clang/libc++ not being [handled properly here](https://github.com/verilator/verilator/blob/1b15509a9ca6c448e9e83d6046664b5c5c79accb/include/verilated_timing.h#L32-L60). Right now I don't have a Mac at hand to take a look. Any help with this would be appreciated.

In the meantime, perhaps you could try different versions of Clang/libc++.

In case I or someone else works on it, what Clang and GCC versions did you test?

HakamAtassi, 2024-08-14:
Hey, 

Thanks again for the response. Yeah, you're right, it does seem to overwrite the std flag when compiling.

Here is some hopefully useful versioning information:

``` ~/ clang --version
Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin```

``` ~/ gcc --version
gcc (Homebrew GCC 14.1.0_2) 14.1.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Thanks for the hint on where to start looking.  I'll see if I can address this over the weekend. Until then, if anyone has any ideas or would like to test anything on a MacBook, I'm happy to be your MacOS proxy. I'll keep this thread updated if I find anything. Thanks again.

JohnnyGOX17, 2024-08-20:
@HakamAtassi I was seeing this same issue today with M1 Macbook running the current Homebrew Verilator version (v5.026). This looks to have been fixed in #5190 though, so I uninstalled the brew version and just built & installed the current `master` branch.

This now works for me, even w/o having to specify `-CFLAGS -std=c++20`. I'd suggest giving that a shot.

HakamAtassi, 2024-08-20:
Hey, 

That's great. The issue seems to be exclusive to the brew install of verilator. Building from source does seem to fix the problem. Thanks again.