verilator/verilator #5364: Allow non-standard 'struct' in VCD traces

VarunKoyyalagunta, 2024-08-13:
I'm requesting that the behavior prior to #4227 be restored under an optional switch.

There are some other proprietary waveform browsers that don't support FSTs and benefit from having `scope struct` in the VCD.

I can create a patch if that is acceptable.

wsnyder, 2024-08-13:
This is non-standard. What tool accepts this, and can you point us to the description of the VCD extensions that tool accepts?

VarunKoyyalagunta, 2024-08-13:
Synopsys's Verdi tool accepts this though I can't find an exact specification. And I believe GtkWave also accepts this, [relevant code](https://github.com/gtkwave/gtkwave/blob/6b18383823a0a7bbb68e9d99b4fc141d012ca103/src/helpers/vcd2fst.c#L722-L816).
Our current workaround is to have verilator dump an fst and do fst -> vcd -> fsdb.

wsnyder, 2024-08-24:
I'm still awaiting a response from Synopsys on what Verdi exactly supports, once we have that we'll decide how to proceed.

wsnyder, 2024-08-28:
Synopsys claims Verdi supports only what is in IEEE, `begin`, `fork`, `function`, `module`, `task`, and anything else is treated same as `module` - but they do not recommend using any other keywords.  Is this inconsistent with what you see?

closed 2024-09-08 by wsnyder

VarunKoyyalagunta, 2024-09-10:
Sorry for my late reply, this slipped by me. Yes so that's inconsistent with what we are seeing. `vcd2fsdb` appears to be treating `struct` differently. With `struct`, in Verdi's nWave window we are able to conveniently expand the struct and see its fields. With `module` that does not work.

wsnyder, 2024-09-10:
Can you please contact your Synopsys FAE then and ask for what is supported?  I don't need to be in the middle & maybe you'll get a better answer ;)

VarunKoyyalagunta, 2024-09-10:
Yeah, let me contact them again, though last time I asked they were also confused. However, if you're okay with reverse engineering it and me trying "struct", "interface", and "union" in verdi, I can do that with likely faster turnaround.

wsnyder, 2024-09-10:
Generally we shouldn't be doing reverse-engineering, better to reach out to them for what is supported.