verilator/verilator #5366: Unrolling Loop Elements in AST.xml

enochliang, 2024-08-13:
Hey. I'm finding a way to unroll the for loops in a SystemVerilog design and **_also unroll the AST elements representing loop_**. So I add a metacomment above the for loop like this (The green part). (The red blocks: loop index)(The yellow blocks: circuit element)
[image]

The picture below is the AST of it. After I add the metacomment, there are still loops in the AST. These part of AST matches the above RTL code.
[image]

QUESTION:
I don't want the <while> block in the AST. _**So I want to know if there is any way I can try to achieve that?**_ 
If not, maybe I can start writing a script to transform the AST.xml.

wsnyder, 2024-08-13:
Run with `--debugi-V3Unroll 9` and it should say why not unrolling.  It cannot unroll if it cannot determine indicies.

wsnyder, 2024-08-24:
Assuming you got this resolved, reopen with what you think is wrong in Verilator if not, thanks.

closed 2024-08-24 by wsnyder