verilator/verilator #5368: Config file effects are non-intuitive in the presence of conflicting directives

gezalore, 2024-08-13:
Just recording this issue for reference, in case somebody runs into it. Not planing on working on it for now.

The effect of config file directives are applied in some arbitrary (but deterministic) order. Furthermore some directive have unalterable effects while others don't. This results in potentially unexpected results when the config file has conflicting instructions (especially through the use of wildcards). None of this seems to be described in the documentation.

For example, one might expect to be able to do this:
```
tracing_off -file "foo_?.sv"  # Don't trace any foo_?.sv - there are many
tracing_on -file "foo_2.sv"  # But actually race foo_2.sv
```

The patterns are checked in lexical order, and because `foo_2` sorts before `foo_?` this will actually still turn off tracing for all foo_? files. We are somewhat lucky that `*` sorts before most characters used in filenames, so the possibly more common use of `*` will do as expected with the current implementation:
```
tracing_off -file "foo_*.sv"
tracing_on -file "foo_2.sv"
```

There are other quirks as well, for example, this will inline everything, as opposed to what one might expect (the inline flag is sticky, see `V3ConfigModule::update`):
```
inline -module "*"
no_inline -module "foo"
```

Argument can be made that the a later directives in the config file should supersede an earlier conflicting one, but anything would be better than undefined behaviour.

wsnyder, 2024-08-13:
#4185 also describes how rules should instead be processed in order.  This write-up is better and not an exact match so I'll leave it open too, though suspect any fix should resolve both issues.

gezalore, 2024-08-13:
FWIW I think it's also not describe whether metacomments take precedence over config files, or command line options.