verilator/verilator #5374: InitArray on non-array internal error when initializing parameter arrays

paul-demo, 2024-08-14:
For both parameters and localparams, this fails in Verilator 5.022:
```systemverilog
localparam BAR = 64;
localparam int FOO [N-1:0] = '{default: BAR};
```

The error message is:
```
%Error: Internal Error: .../V3Width.cpp:2509: InitArray on non-array
```

wsnyder, 2024-08-15:
This works fine on master, please use that or provide an example that fails on master, thanks.

closed 2024-08-15 by wsnyder

paul-demo, 2024-08-15:
Thank you for checking and sorry to waste your time. I traced this to an .svh file which just had `parameter` and `localparam` declarations outside of any module or package definition. That file was getting `include`-ed in the body of a `package`/`endpackage` elsewhere.

Is the recommendation for .svh files which use `include` to just list their containing directories with `-I` and not also list them individually in the list of files passed to verilator? Currently we do both, but clearly this doesn't work when the file itself is not valid verilog syntax when taken out of context.

wsnyder, 2024-08-16:
Common usage is as in C, ypu list only module-containing files on the command line, and headers are included by the module files.

paul-demo, 2024-08-20:
Actually, this still doesn't work even when you **don't** list the .svh file on the command line (and instead only include its directory with the -I flag). Can we add support for this? I think it's somewhat common to have parameters/localparams listed in an .svh file, and other third party tools seem to support it.

wsnyder, 2024-08-21:
Please provide an example that fails on master