verilator/verilator #5376: Performing continous write to wide memory array may trigger a wrong data in the array.

Siudya, 2024-08-15:
# Environment
CPU: AMD Ryzen 7 5700G 
OS: Ubuntu 22.04
CC: clang 18.1.3
verilator: 5.026 2024-06-15 rev v5.026
xmake: 2.9.4

# Reproducing the bug
```
git clone https://github.com/Siudya/qsim
cd qsim
xmake
xmake run emu -d -c 10000
```
The VCD waveform file can be founde in `build/linux/release/sim.vcd`
Actual cmds the scripts run can be shown with
```
xmake -rv
```
Running logs will be similar as followings:
```
Options: Status
seed: 1043
wave_file: sim.vcd
max_cycle: 10000
dump_wave: 0
inject data:
 0xaeef36d95c09cca780f5e0caa3ea1c6b01244ccccc7c1cf5fbf8b928f7f7fff5fe8c30478c184ea138e9f820d5e722d0
inject data:
 0xdd07367040afa67c97b2e51c5e0658319e59f182b94b22f62c81f73e47e4ae1cfa6c0b35bedc620e22f0fa0996f5cf8d
eject data:
 0x5d07367040afa67c97b2e51c5e0658319e59f182b94b22f62c81f73e47e4ae1cfa6c0b35bedc620e22f0fa0996f5cf8d
[0] %Error: QueueSim.sv:38: Assertion failed in TOP.QueueSim: Assertion failed: rx data wrong!
    at QueueSim.scala:72 assert(!rx.io.fault, "rx data wrong!")

%Error: src/rtl/QueueSim.sv:38: Verilog $stop
Aborting...
```
# About the reproducing project
`src/rtl` conatins all DUT files in SystemVerilog.
`src/csrc` contains testbench files in C++.
The `qsim` project consists of a 2-entry FIFO. Each entry of the FIFO has 383 bits width.
`Injector` contains a DPI-C function `queue_inject` to transmit data to the FIFO.
`Ejector` contains a DPI-C function  `queue_eject`  to receive data from FIFO.
All DPI-C functions are implemnted in `src/csrc/myio.cpp`. A scoreboard is also included.
Once the reset is deasserted, `Injector` starts to inject **random** data continuously.

The bug can be obvious when checking write ports of `/TOP/QueueSim/queue/ram_2x383` (W0_*).
A write to addr 0 stores wrong data in `\Memory[0]`.

Moreover, can't figure out if it is my fault, the tx data is not the same as it shows in logs.
In logs, the data is
`0x5d07367040afa67c97b2e51c5e0658319e59f182b94b22f62c81f73e47e4ae1cfa6c0b35bedc620e22f0fa0996f5cf8d`
In waveform, the data is
`0xdd07367040afa67c97b2e51c5e0658319e59f182b94b22f62c81f73e47e4ae1cfa6c0b35bedc620e22f0fa0996f5cf8d`
This may be a flaw in DPI-C implementation.

wsnyder, 2024-08-15:
Unfortunately we are unlikely to have the time to be able to debug such a large test case.  If you reduce it to a minimal test we will take a look.

I'd suggest first try --runtime-debug, and make sure your DPI calls are not exceeding any array bounds.

Siudya, 2024-08-16:
OK I reduced the case to a minimal one on the `reduced` branch of the original repo. I use DPI-C function to directly dirve the array. Run these commands to reproduce:
```
git clone https://github.com/Siudya/qsim -b reduced
cd qsim
xmake
xmake run
```
Run `xmake -rv` and it will show the thorough compilation commands.

The memory array is update at the same clock edge the write enable is asserted. It seems that the outputs of DPI-C functions affect other **sequential** logics immediately,  when a common sensitive clock egde arrives.

wsnyder, 2024-08-18:
DPI routines are not delayed assignments, so a simulator is free to either affect the current or next value. It sounds from your description what you want is this:
  
always @(posedge clock) begin
   reg en_next;
   reg data_next;
   reg addr_next
   dpic_write(en_next, addr_next, data_next, reset_n);
   data <= data_next;
   en <= en_next;
   addr <= addr_next;
end

Siudya, 2024-08-19:
Thanks, it's very clear. I will call dpic functions in negedge instead to avoid this ambiguous situation.

closed 2024-08-19 by Siudya